raspberrypi/pico-sdk #1304: Incorrect PICO_FLASH_SPI_CLKDIV for WaveShare RP2040-Zero

ArkBrj, 2023-03-10:
src\boards\include\boards\waveshare_rp2040_zero.h sets PICO_FLASH_SPI_CLKDIV to 2.
However, according to the board schematics (https://www.waveshare.com/w/upload/4/4c/RP2040_Zero.pdf), this board uses Winbond W25Q16JVUXIQ chip. According to the chip's datasheet (https://www.winbond.com/resource-files/w25q16jv%20spi%20revh%2004082019%20plus.pdf), its max Clock frequency for Read Data instruction (03h) is 50 MHz (Section 10.6 AC Electrical Characteristics). PICO_FLASH_SPI_CLKDIV of 2 sets the clock to around 60 MHz exceeding the spec.
In my experiments with various binaries I see that the code always works right after flashing it, but does not work after restarting the board. This includes the officially released binary for picoprobe.
For home-built projects Debug binaries sometimes work  even after reset but Release do not. Sometimes even Debug binaries do not work after reset.
CircuitPython works fine because it uses PICO_FLASH_SPI_CLKDIV of 4 (confirmed by disassembling the binary).
I think PICO_FLASH_SPI_CLKDIV should be set to 4 in SDK for this board.

AlynxZhou, 2023-06-11:
I meet this problem too with 2 of 3 RP2040-Zeros I bought, a interesting thing is I found setting `PICO_XOSC_STARTUP_DELAY_MULTIPLIER` to `64` also fixes this bug, I think it makes program waits longer before startup.

But your words are more reasonable, and I tried your suggestion, it does fix the bug, so I submit a PR according to this issue. Thanks!

lurch, 2023-06-11:
In case it's useful, see the related discussion in #401

AlynxZhou, 2023-06-11:
> In case it's useful, see the related discussion in #401

Thanks!

My test shows that only change `PICO_FLASH_SPI_CLKDIV` to 4 is enough, there is no need to change both `PICO_FLASH_SPI_CLKDIV` and `PICO_XOSC_STARTUP_DELAY_MULTIPLIER`.

kilograham, 2023-12-15:
merged into develop

closed 2023-12-15 by kilograham

zwenergy, 2025-04-09:
Sorry to bring this old thing back up, but I just stumbled over this. As far as I can see the Flash chip used by the RP2040-Zero board is the same chip used by the regular Pi Pico board (both use the W25Q16JVUXIQ?). So this shouldn't be a direct issue of the Flash chip?

Please correct me if I'm wrong!

Taranus, 2025-07-04:
I agree. In the schematics, both the RP2040-Zero and the official Pico indicate the same reference W25Q16JVUXIQ.
And the marking on the actual chips also state the same (1Pyww). 1P430 on my zero, and 1P113 on my Pico.