raspberrypi/pico-sdk #1306: arm-none-eabi-gcc -v failed

om327, 2023-03-12:
Hi, I'm trying trying to follow the SDK document to link the ARM build chain to vs code in windows. I've manually installed the different elements of the build chain and successfully built all the pico SDK examples. Unfortunately, I can't seem to get vs code to build and receive the following error:

```
[proc] The command: "C:\Program Files\CMake\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug "-DCMAKE_C_COMPILER:FILEPATH=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-gcc.exe" "-DCMAKE_CXX_COMPILER:FILEPATH=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-g++.exe" -SC:/pico/pico-examples -Bc:/pico/pico-examples/build -G "NMake Makefiles" exited with code: 1
[proc] The command: arm-none-eabi-gcc -v failed with error: Error: spawn arm-none-eabi-gcc ENOENT
[proc] The command: arm-none-eabi-g++ -v failed with error: Error: spawn arm-none-eabi-g++ ENOENT
```

I opened vs code from the VS developer command prompt, installed cmake, and updated the ```cmake: configure environment``` and ```cmake: generator```  before loading in the ```pico-examples``` git directory... am i doing something daft?

Many thanks!

om327, 2023-03-12:
so more info from vs code: 

```
CMake Error at CMakeLists.txt:8 (project):The CMAKE_CXX_COMPILER:

  C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.10/bin/arm-none-eabi-g++.exe

is not a full path to an existing compiler tool.

To use the NMake generator with Visual C++, cmake must be run from a shell
that can use the compiler cl from the command line.  This environment is
unable to invoke the cl compiler.  To fix this problem, run cmake from the
Visual Studio Command Prompt (vcvarsall.bat).

Tell CMake where to find the compiler by setting either the environment
variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
to the compiler, or to the compiler name if it is in the PATH.
```

kilograham, 2023-03-12:
please ask help questions here
https://forums.raspberrypi.com/viewforum.php?f=143

closed 2023-03-12 by kilograham