raspberrypi/pico-sdk #1307: `sleep_ms` sometimes stalls until SIGINT from gdb

BtheDestroyer, 2023-03-12:
This is a bit of a followup to #1301

I've got a pair of programs:
- boot3 is in the lower half of flash and RAM and is launched by the stage 2 bootloader
- prog is in the upper half of flash and RAM and is loaded from an SD card and launched by boot3

Everything seems to work perfectly *except* for the fact that prog trying to call `sleep_ms` will get stuck indefinitely in `sleep_until` until I send a SIGINT/halt from gdb, then it immediately realizes enough time has passed and will exit the spin lock. Specifically, every time I halt execution it's at `sleep_until+192` (C source: `uint32_t hi_target = (uint32_t)(target >> 32u);`, ASM: `ldr 42, [sp #20]`)

Is there something I'm missing regarding the RP2040's timers/alarms that I need to un/initialize? Does anyone know why halting from gdb might "wake it up"?

kilograham, 2023-05-26:
seems like a good forum question https://forums.raspberrypi.com/viewforum.php?f=143

but, yeah, gdb halt will wake up the `WFE` instruction and cause it to re-check the timer, so this would suggest that your timer interrupts are not firing for some reason

closed 2023-05-26 by kilograham

jashanbhoora, 2024-07-05:
@BtheDestroyer Did you ever get to the bottom of this? I'm seeing exactly the same thing while running [hello_serial.elf from pico-sdk-examples](https://github.com/raspberrypi/pico-examples/blob/master/hello_world/serial/hello_serial.c) with gdb-multiarch connected via the Raspberry Pi Debug Probe. I get constant "Hello, world!"s as expected when running without gdb.

BtheDestroyer, 2024-07-05:
Nope, never figured out what was going on. Ended up shelving the project I was working indefinitely for unrelated reasons, so it's been a while since I've looked at the situation.

jashanbhoora, 2024-07-05:
Fair enough, thanks for getting back to me!

mstaszenski, 2024-07-05:
Try this:
timer_hw->dbugpause = 0

put on main()

my delays were locking up using debug probe

jashanbhoora, 2024-07-05:
That seems to have done it! Thank you @mstaszenski!

lurch, 2024-07-05:
See also the related discussions at https://github.com/search?q=repo%3Araspberrypi%2Fpico-sdk+dbgpause&type=issues

lsk567, 2024-08-17:
`timer_hw->dbgpause` does the trick for me too. Thanks @mstaszenski and @jashanbhoora!