raspberrypi/pico-sdk #1319: Impossible to install ISR Handler from within library

ventZl, 2023-03-20:
If the code of the application is split into several static and/or object libraries, then it is impossible to install ISR routine from within code provided by any of these libraries. Such routine will get ignored during the link, because `pico-sdk` code is prioritized via the pico_stdlib INTERFACE library.

Consider the following file (isr.c)
~~~~~~~
void SVC_Handler()
{
    // do something
}
~~~~~~~

Then if this code is integrated like following (CMakeLists.txt):
~~~~~~~
add_library(isr STATIC isr.c)
add_executable(hello main.c)
target_link_libraries(hello isr pico_stdlib)
~~~~~~~

Then the `SVC_Handler()` function from within `isr.c` will silently be ignored by the linker (default no-op implementation from pico SDK is used instead). The only possible workaround here is to include the isr.c file as part of the main binary, which may not always be possible/feasible.

kilograham, 2023-03-20:
another possible workaround is to call 

`exception_set_exclusive_handler` from within your lib to do it at runtime

ventZl, 2023-03-21:
Using any such call would make the code highly non-portable.

kilograham, 2023-03-21:
can use use INTERFACE rather than STATIC for your library

ventZl, 2023-03-22:
In this specific case this is not feasible. Static library is used as a handle for precise organization of .data and .bss in memory which is needed for memory protection to be usable. Yet still certain pieces may still want to install ISRs.

Why is the pico_stdlib itself organized as INTERFACE library? That's a highly non-standard approach which almost looks as a hack.