raspberrypi/pico-sdk #1323: Two Pico W's showing up as the same Bluetooth device

thrandale, 2023-03-24:
## Background
I am working on a project where I will want to have multiple Pico W's in the same vicinity, running the same program. The program will be using Bluetooth Low Energy with several services and characteristics.I would Like to be able to give each pico a different name and connect to each one individually.

## The Problem
When I have two picos running and try discovering devices from my phone, I only see one show up. Sometimes it has the name of one pico, and sometimes the name of the other. When I connect to it, sometimes the name matches the one I connect to, and sometimes it is the name of the one I am not connected to.

## Info
- Connecting from An IPhone running IOS 16.3
- Have tried using two different BLE apps to connect, as well as my own custom app (All yielding the same result)
    - nRF Connect
    - LightBlue
    - My own custom app in written in react-native
- Have Restarted the phone to make sure it wasn't using any cached info

## Example
The names are being defined in the `.gatt` file and the `adv_data`.

## Pico 1
#### .gatt file
```
PRIMARY_SERVICE, GAP_SERVICE
CHARACTERISTIC, GAP_DEVICE_NAME, READ, "Laser Car 1"

PRIMARY_SERVICE, GATT_SERVICE
CHARACTERISTIC, GATT_DATABASE_HASH, READ,

PRIMARY_SERVICE, 00000000-6a5c-4ebb-8da6-a4471e0965ef
CHARACTERISTIC, 00000001-6a5c-4ebb-8da6-a4471e0965ef, WRITE_WITHOUT_RESPONSE | DYNAMIC,
CHARACTERISTIC, 00000002-6a5c-4ebb-8da6-a4471e0965ef, WRITE_WITHOUT_RESPONSE | DYNAMIC,
```
### .cpp file
``` cpp
const uint8_t adv_data[] = {
    // Flags general discoverable
    0x02, BLUETOOTH_DATA_TYPE_FLAGS, APP_AD_FLAGS,
    // Name
    0xC, BLUETOOTH_DATA_TYPE_COMPLETE_LOCAL_NAME, 'L', 'a', 's', 'e', 'r', ' ', 'C', 'a', 'r', ' ', '1',
    // Service UUIDs
    0x03, BLUETOOTH_DATA_TYPE_COMPLETE_LIST_OF_16_BIT_SERVICE_CLASS_UUIDS, 0x00, 0x00};
const uint8_t adv_data_len = sizeof(adv_data);
```

## Pico 2
#### .gatt file
```
PRIMARY_SERVICE, GAP_SERVICE
CHARACTERISTIC, GAP_DEVICE_NAME, READ, "Laser Car 2"

PRIMARY_SERVICE, GATT_SERVICE
CHARACTERISTIC, GATT_DATABASE_HASH, READ,

PRIMARY_SERVICE, 00000000-6a5c-4ebb-8da6-a4471e0965ef
CHARACTERISTIC, 00000001-6a5c-4ebb-8da6-a4471e0965ef, WRITE_WITHOUT_RESPONSE | DYNAMIC,
CHARACTERISTIC, 00000002-6a5c-4ebb-8da6-a4471e0965ef, WRITE_WITHOUT_RESPONSE | DYNAMIC,
```
### .cpp file
``` cpp
const uint8_t adv_data[] = {
    // Flags general discoverable
    0x02, BLUETOOTH_DATA_TYPE_FLAGS, APP_AD_FLAGS,
    // Name
    0xC, BLUETOOTH_DATA_TYPE_COMPLETE_LOCAL_NAME, 'L', 'a', 's', 'e', 'r', ' ', 'C', 'a', 'r', ' ', '2',
    // Service UUIDs
    0x03, BLUETOOTH_DATA_TYPE_COMPLETE_LIST_OF_16_BIT_SERVICE_CLASS_UUIDS, 0x00, 0x00};
const uint8_t adv_data_len = sizeof(adv_data);
```

peterharperuk, 2023-03-24:
There's a possible fix for this here https://github.com/raspberrypi/pico-sdk/pull/1284
It sounds like your devices have their mac address unset. Can you tell when and where you got them?
If you run a test in debug, you should see the following message if this is the problem "Warning. No mac in otp. Generating mac from board id"

peterharperuk, 2023-03-25:
Can I ask a favour? When you run your code in debug it should say something like this on the sdio output...

Version: 7.95.61 (abcd531 CY) CRC: 4528a809 Date: Wed 2023-01-11 10:29:38 PST Ucode Ver: 1043.2169 FWID 01-7afb089
cyw43 loaded ok, mac 28:cd:c1:07:86:f7
BT FW download, version = CYW4343A2_001.003.016.0031.0000_Generic_SDIO_37MHz_wlbga_BU_dl_signed
BTstack up and running on 28:CD:C1:07:86:F8.

Can you tell me the mac values your getting on the "cyw43 loaded ok" lines for bother devices?

thrandale, 2023-03-30:
here are the two macs
28:cd:c1:06:2b:3a
28:cd:c1:06:2b:39

Ferdi265, 2023-04-30:
I have another Pico W with a similar case. Here are the MACs I'm observing on 1.5.0:

WiFi MAC: 28:CD:C1:03:C4:BD
Bluetooth MAC: 43:43:A2:12:1F:AC

No warnings about missing the MAC in OTP.

I ordered this Pico W together with a second one only a few weeks ago, which works fine:

WiFi MAC: 28:CD:C1:03:C4:89
Bluetooth MAC: 28:CD:C1:03:C4:8A

The two picos were likely two consecutive picos from one big roll as they were delivered in those little plastic enclosures and the two were still stuck together when I got them.

thrandale, 2023-05-01:
Yes, these were consecutive ones from a roll.

KevinLu9, 2023-05-13:
If you want a temporary fix for the MAC address, you can set the MAC address in code using gap_random_address_set(const bd_addr_t addr) from the BTStack lib.

kilograham, 2023-05-26:
@peterharperuk can you set milestone as appropriate

peterharperuk, 2023-05-30:
Duplicate of https://github.com/raspberrypi/pico-sdk/issues/1269

closed 2023-05-30 by peterharperuk