raspberrypi/pico-sdk #1325: pio_sm_set_jmp_pin() missing

arg08, 2023-03-26:
All of the other PIO configuration features are duplicated between the sm_config_xxx() functions for efficient bulk initialisation and the pio_sm_set_xxx() functions for dynamically tweaking the config.  For example, there's sm_config_set_in_pins() and pio_sm_set_in_pins().

However, pio_sm_set_jmp_pin() is missing (to complement sm_config_set_jmp_pin() which does exist).

I use the following in my code:

```
static inline void pio_sm_set_jmp_pin(PIO pio, uint sm, uint pin)
{
    check_pio_param(pio);
    check_sm_param(sm);
    valid_params_if(PIO, pin < 32);

    pio->sm[sm].execctrl =
        (pio->sm[sm].execctrl & ~PIO_SM0_EXECCTRL_JMP_PIN_BITS)
        | (pin << PIO_SM0_EXECCTRL_JMP_PIN_LSB);
}
```

written in the style of src/rp2_common/hardware_pio/include/hardware/pio.h where it really belongs.

peterharperuk, 2024-08-09:
This has been added

closed 2024-08-09 by peterharperuk

pucgenie, 2024-09-12:
There's code for this procedure in usb_rx.pio.h conflicting with the pio.h one. (Just freshly downloaded and installed the pico_sdk today (on Ubuntu 24.04) as a dependency of GP2040-CE. Can't find the version - grep told me "picotool version 2.0.0")
Deleting the one in usb_rx.pio.h fixed it for me - but how could this happen?

lurch, 2024-09-13:
> EDIT: Sorry, wrong project - bug seems to be inside sekigon-gonnoc/Pico-PIO-USB

Yes, see #1849