raspberrypi/pico-sdk #1326: CFG_TUSB_OS as cmake parameter

rapzak, 2023-03-26:
Hi,

Could you add a cmake parameter to set the CFG_TUSB_OS 

This way we can use the SDK's tinyusb lib in freertos programs as well...

see:
https://www.eevblog.com/forum/microcontrollers/rp2040-how-to-use-freertos-with-tinyusb/

peterharperuk, 2023-03-27:
Doesn't this require a change to tinyusb and not pico-sdk?

peterharperuk, 2023-03-27:
What I mean is, you can already define CFG_TUSB_OS on the cmake command line. The issue is that the tinyusb cmake file won't use it. I guess we can raise a PR against tinyusb to fix this.

rapzak, 2023-03-27:
https://github.com/hathach/tinyusb/discussions/1951

kilograham, 2024-07-20:
@liamfraser just in case you are passing; though enabling it may not be much help until we go and check FreeRTOS support is integrated properly

kilograham, 2026-09-30:
Fixed in 2.0.0 (TinyUSB 0.16): set `TINYUSB_OPT_OS`

closed 2026-09-30 by kilograham