raspberrypi/pico-sdk #1333: [WLAN.status]: Decription of enum value

hasenradball, 2023-03-28:
Dear Support Team,

where can I find the implementation of the `wlan.status()` documented here:
[https://docs.micropython.org/en/latest/library/network.WLAN.html#network.WLAN.status](https://docs.micropython.org/en/latest/library/network.WLAN.html#network.WLAN.status)
```
WLAN.status([param])

    Return the current status of the wireless connection.

    When called with no argument the return value describes the network link status. The possible statuses are defined as constants:
            STAT_IDLE – no connection and no activity,
            STAT_CONNECTING – connecting in progress,
            STAT_WRONG_PASSWORD – failed due to incorrect password,
            STAT_NO_AP_FOUND – failed because no access point replied,
            STAT_CONNECT_FAIL – failed due to other problems,
            STAT_GOT_IP – connection successful.
 ```
 Because the assumption, that the list is ordered in ascending order of the status value seems not the case, right?
Or where can I look up the values of the enums?

kilograham, 2023-03-28:
MicroPython issues belong here https://github.com/micropython/micropython

closed 2023-03-28 by kilograham

hasenradball, 2023-03-29:
@kilograham 
Hi can you give me an hint were to find the wlan related sources?

hasenradball, 2023-03-29:
Sorry I seems something was going wrong.
I would thing I wrote it in the micropython part.