raspberrypi/pico-sdk #1335: Feature request: pico-stdio: Provide `puts_raw` analogue that doesn't write newline

cocasema, 2023-03-29:
It would be great to have access to `stdio_put_string()` version that doesn't append newline at the end - that would simplify logging and similar routines, and can help in some cases with custom stdio drivers.

``` c
int putchar_raw(int c) {
    char cc = (char)c;
    stdio_put_string(&cc, 1, false, true);
    return c;
}

int puts_raw(const char *s) {
    int len = (int)strlen(s);
    stdio_put_string(s, len, true, true);
    stdio_flush();
    return len;
}

//-- suggestion --//
int put_string_raw(const char *s) {
    int len = (int)strlen(s);
    stdio_put_string(s, len, false, true);
    stdio_flush();
    return len;
}
```

If that's acceptable, I can send a PR.

kilograham, 2023-05-26:
seems ok, but name is probably confusing maybe `writes_raw` is a better name

peterharperuk, 2024-08-09:
putchar_raw and puts_raw have been added. Marking this as resolved.

cocasema, 2024-08-09:
> putchar_raw and puts_raw have been added. Marking this as resolved.

I'm a bit confused. Both `putchar_raw` and `puts_raw` were already there. The request was to add `writes_raw` (`put_string_raw`).

peterharperuk, 2024-08-09:
You're right. My misunderstanding. Lets see what @kilograham says.

kilograham, 2024-08-09:
fixed

kilograham, 2026-09-30:
Fixed in 2.0.0: `stdio_put_string()` is now public

closed 2026-09-30 by kilograham