raspberrypi/pico-sdk #1337: async_at_time_worker with default init on core1, alarm callback asserts

slimhazard, 2023-04-01:
I tried to use an `async_at_time_worker` in an app that links `pico_cyw43_arch_lwip_threadsafe_background`, with all of the network and async code initialized and running solely on core1. The first attempt was with default initialization of the async context -- no explicit invocation of `async_context_threadsafe_background_init()`, so that `cyw43_arch_init()` creates the context implicitly.
```
/* No previous init of async_context. All on core1 */
cyw43_arch_init_with_country(country);
async_context_add_at_time_worker_in_ms(cyw43_arch_async_context(), &led_indicator, TIMEOUT_MS);
```
The alarm pool callback panicked:
```
assertion "id && id <= pool->heap->max_nodes" failed: file "[...]/pico-sdk/src/common/pico_time/time.c", line 65, function: get_entry_id_high

(gdb) bt
[...]
#2  0x1000866a in get_entry_id_high (id=0 '\000', pool=0xffffffff)
    at [...]/pico-sdk/src/common/pico_time/time.c:65
#3  alarm_pool_alarm_callback (alarm_num=268450719)
    at [...]/pico-sdk/src/common/pico_time/time.c:147
[...]

(gdb) f 2
#2  0x1000866a in get_entry_id_high (id=0 '\000', pool=0xffffffff)
    at [...]/pico-sdk/src/common/pico_time/time.c:65
65	    assert(id && id <= pool->heap->max_nodes);

(gdb) p id
$1 = 0 '\000'
(gdb) p pool->heap->max_nodes 
$2 = 49 '1'
```
The assertion triggers because `id` == 0.

In [this Pico forum thread](https://forums.raspberrypi.com/viewtopic.php?t=349613), @peterharperuk gave some hints about initializing the async_context explicitly. I suspected that the alarm pool needs to be set in its configuration, since the default alarm pool is not available on core1. And in fact this works:
```
void
core1_main(void)
{
	/* At initialization time in the core1 main function. */
	async_context_threadsafe_background_t ctx;
	alarm_pool_t *pool;
	async_context_threadsafe_background_config_t cfg;

	pool = alarm_pool_create_with_unused_hardware_alarm(MAX_TIMERS);
	cfg = async_context_threadsafe_background_default_config();
	cfg.custom_alarm_pool = pool;
	async_context_threadsafe_background_init(&ctx, &cfg);

	/* This is the "forever loop" on core1. */
	for (;;)
		run( (async_context_t *)&ctx );
}

/* The run function called in the forever loop */
static void
run(async_context_t *ctx)
{
	/* ... */

	cyw43_arch_set_async_context(ctx);
	if ((err = cyw43_arch_init_with_country(country)) != PICO_OK) {
		/* ... error handling ... */
	}

	/* ... */

	async_context_add_at_time_worker_in_ms(ctx, &led_indicator, TIMEOUT_MS);

	/* ... */
}
```
The at_time_worker then functions as expected without assertion (the LED flashes as I wanted it to).

But the [documentation of the custom_alarm_pool member](https://www.raspberrypi.com/documentation/pico-sdk/structasync__context__threadsafe__background__config.html#a8776bae13027752f07a4f3170b2c47b2) of `struct async_context_threadsafe_background_config` seems to be saying that default initialization should have set up the alarm pool on core1 correctly:

> The default alarm pool used is the "default alarm pool" (see alarm_pool_get_default()) if available, and if that is on the same core, otherwise a private alarm_pool instance created during initialization. 

@kilograham said in the forum thread that under these conditions, this may have been a bug -- the network stack runs solely on core1, `cyw43_arch_init()` with its implicit async_context init called on core1, and the alarm pool runs on core1. Hence the github issue.

kilograham, 2023-06-02:
i tried this on `develop` branch right now (well something equivalent code wise), and it worked fine... please re-open if you can still reproduce on `develop`

closed 2023-06-02 by kilograham

slimhazard, 2023-06-06:
Can confirm that it's working on `develop`.

That is, that it suffices to call `cyw43_arch_async_context()`, without having to explicitly get an unused alarm pool and then call `async_context_threadsafe_background_init()`.