raspberrypi/pico-sdk #1339: Arbitrary string length hard coded in tud_descriptor_string_cb() in stdio_usb_descriptors.c

hubiscode, 2023-04-02:
The length of USB descriptor strings is limited to 20 characters through a define statement in the descriptor call back:

```
const uint16_t *tud_descriptor_string_cb(uint8_t index, __unused uint16_t langid) {
    #define DESC_STR_MAX (20)
    static uint16_t desc_str[DESC_STR_MAX];
```

Since version 1.5, it is possible to supply custom manufacturer and product strings by defining USBD_MANUFACTURER and USBD_PRODUCT, but they get truncated at 20 characters.  It would be nice to permit a user-defined maximum string length, e.g.,  by defining USBD_DESC_STR_MAX and changing the call back code to

```
const uint16_t *tud_descriptor_string_cb(uint8_t index, __unused uint16_t langid) {
#ifndef USBD_DESC_STR_MAX
#define USBD_DESC_STR_MAX (20)
#elif USBD_DESC_STR_MAX > 127
#error USBD_DESC_STR_MAX too high (max is 127).
#elif USBD_DESC_STR_MAX < 16
#error USBD_DESC_STR_MAX too low (min is 16).
#endif
    static uint16_t desc_str[USBD_DESC_STR_MAX];
```

The preprocessor statements above include range checking for the maximum descriptor length so it is always long enough to return the serial number and not longer than the maximum possible length.

hubiscode, 2023-04-05:
I created a pull request and fixed the bounds check. The lower bound is 17 (16 for the serial number plus 1 header).

kilograham, 2023-05-29:
merged into develop

PR #1341 (2023-05-29): Customizable USB descriptor string length (Fixes #1339)
Implement a mechanism to set the maximum string length used in tud_descriptor_string_cb() by defining USBD_DESC_STR_MAX. If USBD_DESC_STR_MAX is not defined, the behavior remains unchanged and the previous default value of 20 is used. A compile time error is produced if USBD_DESC_STR_MAX is higher than 127 since the length of the string plus header is returned in a single byte as (2 * len + 2). Similarly, a compile time error is generated if the length is defined as less than 17 in order to ensure that there is enough room for the 16-char [... cut]
closed 2023-05-29 by kilograham