raspberrypi/pico-sdk #1343: c++20 warning fix

GFoxPM, 2023-04-17:
warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
[/src/rp2_common/cmsis/stub/CMSIS/Core/Include/mpu_armv7.h#L210](https://github.com/raspberrypi/pico-sdk/blob/2ccab115de0d42d31d6611cca19ef0cd0d2ccaa7/src/rp2_common/cmsis/stub/CMSIS/Core/Include/mpu_armv7.h#L210)
```diff
-  MPU->CTRL  &= ~MPU_CTRL_ENABLE_Msk;
+  MPU->CTRL  = MPU->CTRL & ~MPU_CTRL_ENABLE_Msk;
```

kilograham, 2024-07-20:
@will-v-pi is this fixed with latest CMSIS headers?

will-v-pi, 2024-07-23:
No, although I think it's being looked into - see https://github.com/ARM-software/CMSIS_5/issues/1544 and https://github.com/ARM-software/CMSIS_6/issues/121

These CMSIS stubs are taken directly from the upstream CMSIS repo, so we won't be modifying them in this repo. Once they're fixed there we'll update our headers here

will-v-pi, 2025-07-04:
No, this issue with the CMSIS stubs has not been fixed upstream - I'm not sure if they're planning on fixing it either

closed 2025-07-07 by kilograham