raspberrypi/pico-sdk #1344: pico/stdlib.h not found

zaze06, 2023-04-17:
Hi, So I'm trying to start developing for pi pico but I'm unable to get the environment setup

I followed the steps that were provided in the [readme](https://github.com/raspberrypi/pico-sdk#quick-start-your-own-project)

I'm running and developing on Windows 11 if that's necessary information

So my main.c looks like this
```c
#include <stdio.h>
#include "pico/stdlib.h"

int main() {
    setup_default_uart();
    printf("Hello, world!\n");
    return 0;
}
```

my CMakeLists.txt
```cmake
cmake_minimum_required(VERSION 3.13)

include(pico_sdk_import.cmake)

project(display C)

pico_sdk_init()

set(CMAKE_C_STANDARD 23)

add_executable(display main.c)

# Add pico_stdlib library which aggregates commonly used features
target_link_libraries(display pico_stdlib)

# create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(display)
```

<details>
  <summary>Cmake log</summary>
  
  ```txt
  PICO_SDK_PATH is C:/Users/<my user>/clib/pico
  PICO platform is rp2040.
  Build type is Debug
  Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
  PICO target board is pico.
  Using board configuration from C:/Users/<my user>/clib/pico/src/boards/include/boards/pico.h
  TinyUSB available at C:/Users/<my user>/clib/pico/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
  Compiling TinyUSB with CFG_TUSB_DEBUG=1
  BTstack available at C:/Users/<my user>/clib/pico/lib/btstack
  cyw43-driver available at C:/Users/<my user>/clib/pico/lib/cyw43-driver
  Pico W Bluetooth build support available.
  lwIP available at C:/Users/<my user>/clib/pico/lib/lwip
  mbedtls available at C:/Users/<my user>/clib/pico/lib/mbedtls
  -- Configuring done
  CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
  Missing variable is:
  CMAKE_ASM_COMPILE_OBJECT
  CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
  Missing variable is:
  CMAKE_CXX_COMPILE_OBJECT
  -- Generating done
  CMake Generate step failed.  Build files cannot be regenerated correctly.
  
  [Failed to reload]
  ```
 </details>
 
 Thanks in advance and I apologize if this has already been solved before. I'm submitting this since I was unable to find a solution

peterharperuk, 2023-04-17:
Can you try changing the project line to...

project(display C CXX ASM)

zaze06, 2023-04-17:
That seems to have done it. thank you.