raspberrypi/pico-sdk #1352: CMake 3.13 supports sys/poll.h but reports "No such file or directory" when building

zpotts-cnu, 2023-04-25:
When including sys/poll.h in C, make reports a fatal error:
"fatal error: sys/poll.h: No such file or directory"

When using:
`#include <sys/poll.h`

I have also tried:
`#include <poll.h>`

Any help would be appreciated!

I am developing in Ubuntu 20.04 and Raspberry Pi OS for the respective project.

peterharperuk, 2023-04-25:
I don't really know what you mean by "CMake 3.13 supports sys/poll.h". The sdk doesn't have any concept of a file descriptor

zpotts-cnu, 2023-04-25:
> I don't really know what you mean by "CMake 3.13 supports sys/poll.h". The sdk doesn't have any concept of a file descriptor

Thank you for getting back to me. I apologize, as my familiarity with file descriptors and io on the pico is limited.

Also, I understand I am about to get off topic in regard to a pico-sdk issue. I completely understand if this cannot move forward. Thank you!

I am trying to buffer communication between a Raspberry Pi and a PICO over UART connection via the USB bus. It started as the code seen below but I wanted to add a timeout or essentially check if there is anything to read since scanf is blocking and waits until something is received. Instructions also come in blocks so I wanted to read until the block is done.

```
void read_stream(char* buff) {
    char instr[CMD_LEN];

    scanf("%8s", instr);
    enqueue(instr, buff);  // This is a function I defined to buffer the instr into buff
    printf("Data received: %s\n", instr);
}
```

I attempted to use poll:

```
void read_stream(char* buff) {
    char instr[CMD_LEN];

    struct pollfd stdin_poll = {STDIN_FILENO, POLLIN|POLLPRI};  // Connection to stdin file descriptor defining instream signal
    if (poll(&stdin_poll, 1, 1000)) {  // Check stdin for data, timeout after 1 second
        while(strcmp(instr, stall_instr) != 0) {  // While we don't get a stall
            scanf("%8s", instr);
            enqueue(instr, buff);
            printf("Data received: %s\n", instr);
        }
    }
}
```

This is what I have landed on temporarily (I am testing it today):

```
void read_stream(char* buff) {
    char instr[CMD_LEN];

    fd_set readfds;
    int num_readable;
    struct timeval tv;
    int fd_stdin = fileno(stdin);

    FD_ZERO(&readfds);
    FD_SET(fileno(stdin), &readfds);

    tv.tv_sec = 1;
    tv.tv_usec = 0;

    num_readable = select(fd_stdin + 1, &readfds, NULL, NULL, &tv);
    if (num_readable == 1) {
        while(strcmp(instr, stall_instr) != 0) {  // While we don't get a stall
            scanf("%8s", instr);
            enqueue(instr, buff);
            printf("Data received: %s\n", instr);
        }
    }
}
```

What do you recommend? I am about to start stepping through solutions. Unfortunately, one is tossing the IO on the other core so that is blocked instead of the main core.

lurch, 2023-04-25:
Does #1213 do what you're looking for?

peterharperuk, 2023-04-25:
Are you having trouble with the code running on the Raspberry Pi? That runs Linux usually and so has a kernel that can manage file descriptors. Unfortunately this code won't work on a Pico and you'll need something a bit lower level. There are some examples in pico-examples.

(@lurch I think https://github.com/raspberrypi/pico-sdk/pull/1213 is only really relevant to sdio?)

kilograham, 2023-05-26:
not really sure what the question/issue is here

closed 2023-05-26 by kilograham