raspberrypi/pico-sdk #1354: Slow and large installation due to unnecessary TinyUSB components

Liftyee, 2023-04-26:
When installing the pico-sdk, the installation takes much longer than it should because many unnecessary drivers from the TinyUSB repo are cloned individually due to the references. [https://github.com/hathach/tinyusb/tree/86c416d4c0fb38432460b3e11b08b9de76941bf5/hw/mcu](Example of folder that is slowing down the installation) - surely all the infineon, wch, etc. drivers are not needed? 

The pico-sdk could be made much smaller and quicker to install if only the required parts of TinyUSB were included.

dhalbert, 2023-04-26:
TinyUSB recently removed submodules. See https://github.com/hathach/tinyusb/pull/1947 and https://github.com/hathach/tinyusb/pull/1961. So maybe this problem will go away when TinyUSB is updated?

Tagging @hathach for interest.

Liftyee, 2023-04-26:
Looks like [hathach/tinyusb#1947](https://github.com/hathach/tinyusb/pull/1947) will improve install times, although hopefully the python script won't break anything. 

Leaving issue open until that PR is merged.

lurch, 2023-04-26:
See also #1044 

You only get these "unnecessary drivers" if you do a recursive clone, which **isn't** what https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf says to do (for precisely this reason :wink: ). I'm curious which instructions you were following which means you _are_ pulling down all the submodules?