raspberrypi/pico-sdk #1355: Pioasm does not fail if there are more than 32 instructions divided into two state machines.

MrGreensWorkshop, 2023-04-27:
Hi, as the title explains.

As datasheet says, we can use 32 instructions per PIO.
If we try to compile a program that has more than 32 instructions in a single state machine, we get the error below, as expected.
```
test.pio:60.1-3: program instruction limit of 32 instruction(s) exceeded
```
When I'm using two state machines (or more) in a single PIO, let's say one has 16 instructions and the second has 20, I don't get a compile error. I think Pioasm doesn't check the total count of instructions for all state machines per PIO. When I write the binary to the Raspberry Pi Pico, it doesn't work and hangs, which took me 6 hours to figure out why.

rbehm-ibb, 2023-04-27:
Fail would be bad since it is possible that one does not use all SM simultaneously. You need  to check the result of  pio_add_program(...)

lurch, 2023-04-27:
Also `pio_can_add_program`
https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#gaa8aafe966a46f4de9be59f73cae65a4b

MrGreensWorkshop, 2023-04-27:
> Fail would be bad since it is possible that one does not use all SM simultaneously. You need to check the result of pio_add_program(...)

It doesn't matter if it is used simultaneously or not; what does matter is the size of the instruction count allowed per pio.

@peterharperuk, @lurch 
When people see this error below, thinks compiler checks it. But its not. 
```
test.pio:60.1-3: program instruction limit of 32 instruction(s) exceeded
```
May I ask, is there any Pioasm documentation explaining what's to be checked or not? I don't think it's nice.

I'm just saying, tons of people gonna waste time on this..

peterharperuk, 2023-04-27:
You've lost me. Maybe I'm misunderstanding your problem. Programs are compiled and it correctly complains if you use more than 32 instructions. You could load two 32 instructions programs into the two pios and it would presumably work. If you try and load "at runtime" two programs into one pio that causes it to exceed 32 instructions then it will fail... at runtime. I don't know how a compiler can be expected to check this. You might compile any number of programs and just load them "at runtime" as and when needed.

MrGreensWorkshop, 2023-04-27:
I see. I guess I didn't understand well.

Could you please explain "loading at runtime" or " not loading at runtime"?
Because I can only see one way, "loading at runtime" with the pio_add_program function

>You might compile any number of programs and just load them "at runtime" as and when needed.

I miss this part. So what I understand is that we can have two programs that exceed 32 instructions, as long as they are not loaded by pio_add_program. Am I correct?

```
.program test1
;20 instruction

.program test2
;20 instruction
```

Since I only can find "pio_can_add_program" function only one example blow. 
https://github.com/raspberrypi/pico-examples/blob/master/pio/ir_nec/nec_transmit_library/nec_transmit.c

I think its better this is documented somewhere here. (right now there is nothing about pio_can_add_program)
https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf
https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

Sorry for taking your time. Please close the issue.

peterharperuk, 2023-04-27:
I think you've understood now - the tool just compiles the program. To actually use it you have to call pio_add_program. You could stop using it and load something else with pio_remove_program. The API docs are around here https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#ga3571e3c7c6e645e92b8d438203a911b9

The best example is probably what's done for PICO_W, see cyw43_spi_init in the sdk. It uses pio_can_add_program to find which of the pios has space for the program, then pio_claim_unused_sm to claim a state machine on that pio. Finally it calls pio_add_program which returns an offset in the memory where the program is stored. I wouldn't say it's very easy to use - we've had to fix a few bugs in there. We could probably do with a helper function to implement this logic to make it easier.

MrGreensWorkshop, 2023-04-27:
@peterharperuk Thanks. I've just added "pio_can_add_program" to my program, and now I can see whether it's ok or not.

I still think it should be documented. Something like this.

You can have multiple Pioasm programs; since Pio doesn't have flash memory, its programs are held in flash memory, which you can load into instruction memory using pio_add_program. Since you can load multiple programs into instruction memory, please make sure there is enough space in instruction memory using the pio_can_add_program function.

peterharperuk, 2023-04-27:
I would suggest raising an issue against the documentation if you think unclear https://github.com/raspberrypi/documentation/

closed 2023-04-27 by peterharperuk

aallan, 2023-04-27:
> I would suggest raising an issue against the documentation if you think unclear https://github.com/raspberrypi/documentation/

Issues around the PDF documentation should go to the [pico-feedback](https://github.com/raspberrypi/pico-feedback/issues) repo.

MrGreensWorkshop, 2023-04-27:
@aallan 
I open the issue there, https://github.com/raspberrypi/documentation/issues/2902

@peterharperuk 
Thank you for your time.

aallan, 2023-04-27:
I transferred it from the documentation to the pico-feedback repo.