raspberrypi/pico-sdk #1358: i have problem trying to use #include "pico/binary_info.h"

FJay81, 2023-04-28:
`[ 26%] Building C object CMakeFiles/led.dir/C_/VSARM/sdk/pico/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
[ 27%] Building C object CMakeFiles/led.dir/C_/VSARM/sdk/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj
[ 28%] Building C object CMakeFiles/led.dir/C_/VSARM/sdk/pico/pico-sdk/src/rp2_common/pico_platform/platform.c.obj
In file included from C:\VSARM\sdk\pico\pico-sdk\src\common\pico_binary_info\include/pico/binary_info.h:28,
                 from C:\Users\name\OneDrive\Desktop\code\microcontroller\pico\picoW\wifi tool\ctest\main.c:5:
C:\Users\name\OneDrive\Desktop\code\microcontroller\pico\picoW\wifi tool\ctest\main.c: In function 'main':
C:\VSARM\sdk\pico\pico-sdk\src\common\pico_binary_info\include/pico/binary_info/structure.h:107:32: error: initializer element is not constant
  107 | #define BI_PINS_ENCODING_MULTI 2
      |                                ^
C:\VSARM\sdk\pico\pico-sdk\src\common\pico_binary_info\include/pico/binary_info/code.h:35:44: note: in definition of macro 'bi_decl'
   35 | #define bi_decl(_decl) __bi_mark_enclosure _decl; __bi_decl(__bi_ptr_lineno_var_name, &__bi_lineno_var_name.core, ".binary_info.keep.", __used);
      |                                            ^~~~~
C:\VSARM\sdk\pico\pico-sdk\src\common\pico_binary_info\include/pico/binary_info/code.h:133:54: note: in expansion of macro '__bi_encoded_pins_with_func'
  133 | #define bi_3pins_with_func(p0, p1, p2, func)         __bi_encoded_pins_with_func(BI_PINS_ENCODING_MULTI | ((func << 3)) | ((p0) << 7) | ((p1) << 12) | ((p2) << 17) | ((p2) << 22))
      |                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\VSARM\sdk\pico\pico-sdk\src\common\pico_binary_info\include/pico/binary_info/code.h:133:82: note: in expansion of macro 'BI_PINS_ENCODING_MULTI'
  133 | #define bi_3pins_with_func(p0, p1, p2, func)         __bi_encoded_pins_with_func(BI_PINS_ENCODING_MULTI | ((func << 3)) | ((p0) << 7) | ((p1) << 12) | ((p2) << 17) | ((p2) << 22))
      |                                                                                  ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\name\OneDrive\Desktop\code\microcontroller\pico\picoW\wifi tool\ctest\main.c:38:13: note: in expansion of macro 'bi_3pins_with_func'
   38 |     bi_decl(bi_3pins_with_func(MISO, MOSI, CLK, GPIO_FUNC_SPI));
      |             ^~~~~~~~~~~~~~~~~~
mingw32-make: *** [all] Error 2
PS C:\Users\name\OneDrive\Desktop\code\microcontroller\pico\picoW\wifi tool\ctest\build>

` terminal output

my cmake
[CMakeLists.txt](https://github.com/raspberrypi/pico-sdk/files/11357470/CMakeLists.txt)

the code
[main.txt](https://github.com/raspberrypi/pico-sdk/files/11357472/main.txt)

Reneg973, 2023-04-29:
you didn't declare your pins as constants:

`uint CS = 17;
uint MOSI = 19;
uint MISO = 16;
uint CLK = 18;`

kilograham, 2023-04-29:
Please ask questions here https://forums.raspberrypi.com/viewforum.php?f=144

however these need to be compile time constants to be included in binary info (so use `#define`)

closed 2023-04-29 by kilograham