raspberrypi/pico-sdk #1360: Build a project with a different build system

tmayoff, 2023-04-30:
I know this is probably not very supported so I don't expect much, but I was hoping to be able to develop using the meson build system as I'm a lot more familiar with it. Meson has a module to build cmake subprojects and was hoping that I'd be able to get it working however I've not had any luck.

This is the meson file

```meson.build
project('blinky', 'cpp', version: '0.1')

cmake = import('cmake')
pico_sdk_opts = cmake.subproject_options()
pico_sdk_opts.add_cmake_defines({'PICO_BOARD': 'pico_w'})
pico_sdk_proj = cmake.subproject('pico_sdk', options: pico_sdk_opts)
```

```
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/tyler/src/blinky/build
ninja: Entering directory `/home/tyler/src/blinky/build'
[0/1] Regenerating build files.
The Meson build system
Version: 1.0.1
Source dir: /home/tyler/src/blinky
Build dir: /home/tyler/src/blinky/build
Build type: native build
Project name: blinky
Project version: 0.1
C++ compiler for the host machine: /usr/bin/g++ (gcc 13.1.1 "g++ (GCC) 13.1.1 20230426 (Red Hat 13.1.1-1)")
C++ linker for the host machine: /usr/bin/g++ ld.bfd 2.39-9
Host machine cpu family: x86_64
Host machine cpu: x86_64

Executing subproject pico_sdk method cmake 

pico_sdk| Found CMake: /usr/bin/cmake (3.26.3)

| Configuring the build directory with CMake version 3.26.3
| Running CMake with: -G Ninja -DCMAKE_INSTALL_PREFIX=/usr/local -DPICO_BOARD=pico_w
|   - build directory:          /home/tyler/src/blinky/build/subprojects/pico_sdk/__CMake_build
|   - source directory:         /home/tyler/src/blinky/subprojects/pico_sdk
|   - toolchain file:           /home/tyler/src/blinky/build/subprojects/pico_sdk/__CMake_build/CMakeMesonToolchainFile.cmake
|   - preload file:             /usr/lib/python3.11/site-packages/mesonbuild/cmake/data/preload.cmake
|   - trace args:               --trace-expand --trace-format=json-v1 --no-warn-unused-cli --trace-redirect=cmake_trace.txt
|   - disabled policy warnings: [CMP0025, CMP0047, CMP0056, CMP0060, CMP0065, CMP0066, CMP0067, CMP0082, CMP0089, CMP0102]

| CMake Warning (dev) at pico_sdk_init.cmake:36 (set):
| Policy CMP0126 is not set: set(CACHE) does not remove a normal variable of
| the same name.  Run "cmake --help-policy CMP0126" for policy details.  Use
| the cmake_policy command to set the policy and suppress this warning.

| For compatibility with older versions of CMake, normal variable
| "PICO_SDK_PATH" will be removed from the current scope.
| Call Stack (most recent call first):
| CMakeLists.txt:8 (include)
| This warning is for project developers.  Use -Wno-dev to suppress it.
| Running with expanded trace output on.
| Not searching for unused variables given on the command line.
| Trace will be written to cmake_trace.txt

| PICO_SDK_PATH is /home/tyler/src/blinky/subprojects/pico_sdk
| Defaulting PICO_PLATFORM to rp2040 since not specified.
| CMake Warning (dev) at cmake/pico_pre_load_platform.cmake:14 (set):
| Policy CMP0126 is not set: set(CACHE) does not remove a normal variable of
| the same name.  Run "cmake --help-policy CMP0126" for policy details.  Use
| the cmake_policy command to set the policy and suppress this warning.

| For compatibility with older versions of CMake, normal variable
| "PICO_PLATFORM" will be removed from the current scope.
| Call Stack (most recent call first):
| pico_sdk_init.cmake:45 (include)
| CMakeLists.txt:8 (include)
| This warning is for project developers.  Use -Wno-dev to suppress it.

| -- Defaulting build type to 'Release' since not specified.
| -- The C compiler identification is GNU 13.1.1
| -- The CXX compiler identification is GNU 13.1.1
| -- The ASM compiler identification is GNU
| -- Found assembler: /usr/bin/gcc
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /usr/bin/gcc - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /usr/bin/g++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| Build type is Release
| PICO target board is pico_w.
| Using CMake board configuration from /home/tyler/src/blinky/subprojects/pico_sdk/src/boards/pico_w.cmake
| Using board configuration from /home/tyler/src/blinky/subprojects/pico_sdk/src/boards/include/boards/pico_w.h
| CMake Warning (dev) at src/rp2_common/boot_stage2/CMakeLists.txt:21 (set):
| Policy CMP0126 is not set: set(CACHE) does not remove a normal variable of
| the same name.  Run "cmake --help-policy CMP0126" for policy details.  Use
| the cmake_policy command to set the policy and suppress this warning.

| For compatibility with older versions of CMake, normal variable
| "PICO_DEFAULT_BOOT_STAGE2" will be removed from the current scope.
| This warning is for project developers.  Use -Wno-dev to suppress it.

| -- Found Python3: /usr/bin/python3.11 (found version "3.11.3") found components: Interpreter
| TinyUSB available at /home/tyler/src/blinky/subprojects/pico_sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
| CMake Warning (dev) at lib/tinyusb/hw/bsp/rp2040/pico_sdk_import.cmake:60 (set):
| Policy CMP0126 is not set: set(CACHE) does not remove a normal variable of
| the same name.  Run "cmake --help-policy CMP0126" for policy details.  Use
| the cmake_policy command to set the policy and suppress this warning.

| For compatibility with older versions of CMake, normal variable
| "PICO_SDK_PATH" will be removed from the current scope.
| Call Stack (most recent call first):
| lib/tinyusb/hw/bsp/rp2040/family.cmake:11 (include)
| lib/tinyusb/hw/bsp/family_support.cmake:134 (include)
| src/rp2_common/tinyusb/CMakeLists.txt:24 (include)
| This warning is for project developers.  Use -Wno-dev to suppress it.

| CMake Warning (dev) at lib/tinyusb/hw/bsp/family_support.cmake:141 (set):
| Policy CMP0126 is not set: set(CACHE) does not remove a normal variable of
| the same name.  Run "cmake --help-policy CMP0126" for policy details.  Use
| the cmake_policy command to set the policy and suppress this warning.

| For compatibility with older versions of CMake, normal variable
| "FAMILY_MCUS" will be removed from the current scope.
| Call Stack (most recent call first):
| src/rp2_common/tinyusb/CMakeLists.txt:24 (include)
| This warning is for project developers.  Use -Wno-dev to suppress it.

| BTstack available at /home/tyler/src/blinky/subprojects/pico_sdk/lib/btstack
| cyw43-driver available at /home/tyler/src/blinky/subprojects/pico_sdk/lib/cyw43-driver
| Pico W Bluetooth build support available.
| lwIP available at /home/tyler/src/blinky/subprojects/pico_sdk/lib/lwip
| Pico W Wi-Fi build support available.
| mbedtls available at /home/tyler/src/blinky/subprojects/pico_sdk/lib/mbedtls
| CMake Warning (dev) at CMakeLists.txt:48 (set):
| Policy CMP0126 is not set: set(CACHE) does not remove a normal variable of
| the same name.  Run "cmake --help-policy CMP0126" for policy details.  Use
| the cmake_policy command to set the policy and suppress this warning.

| For compatibility with older versions of CMake, normal variable
| "PICO_SDK_TESTS_ENABLED" will be removed from the current scope.
| This warning is for project developers.  Use -Wno-dev to suppress it.

| -- Configuring done (1.4s)
| -- Generating done (0.1s)
| -- Build files have been written to: /home/tyler/src/blinky/build/subprojects/pico_sdk/__CMake_build

pico_sdk| CMake configuration: SUCCEEDED
pico_sdk| WARNING: Unable to evaluate the cmake variable '$<TARGET_FILE:bs2_default>'.

../meson.build:7:0: ERROR: We evaluated the cmake variable '' to an empty string, which is not a valid path to an executable.

A full log can be found at /home/tyler/src/blinky/build/meson-logs/meson-log.txt
FAILED: build.ninja 
/usr/bin/meson --internal regenerate /home/tyler/src/blinky /home/tyler/src/blinky/build
ninja: error: rebuilding 'build.ninja': subcommand failed
```

If there's anyone familiar with both systems that could help I'd really appreciate it.

Kloenk, 2023-05-02:
As far as I can tell, the pico-sdk has both parts of a cmake library, and a code library. I guess that would have to be splited, and then the main CMakeLists.txt has to only build the code objects, and you have to recreate the cmake lib part in meson

marek22k, 2025-06-06:
Are there any news? For reasons of simplicity, I strongly prefer meson.

d-c-d, 2025-06-06:
Is `PICO_DEFAULT_BOOT_STAGE2_FILE` coming from an environment variable?
it appears that you are not getting the boot stage2 loader set correctly
You are targeting `rp2040`  which needs the boot loader