raspberrypi/pico-sdk #1363: pio_disassembler.h missing include

MrJake222, 2023-05-06:
After Arch update to GCC 13 `pio_disassembler.h` won't build:

```
/usr/share/pico-sdk/tools/pioasm/pio_disassembler.h:16:25: error: ‘uint16_t’ was not declared in this scope
   16 | std::string disassemble(uint16_t inst, uint sideset_bits, bool sideset_opt);
      |                         ^~~~~~~~
/usr/share/pico-sdk/tools/pioasm/pio_disassembler.h:1:1: note: ‘uint16_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
  +++ |+#include <cstdint>
```

kilograham, 2023-05-06:
Did you try 1.5.1 already?

yes, the include should be in `pio_disassembler.h` not `pio_disassembler.cpp`, so you could undo #1317 / #138 fix as part of this PR if you like

MrJake222, 2023-05-07:
This is based on `master` branch so it's not possible.

> Did you try 1.5.1 already?

You mean `develop` branch?

lurch, 2023-05-07:
I think @kilograham was suggesting that in **your** PR #1364 you should probably also "undo the effect" of #1318 at the same time? i.e. remove the `#include <cstdint>` from `pio_disassembler.cpp`

> This is based on master branch so it's not possible.

I'm not sure what that was referring to, but all PRs to this repo need to be against `develop`, see https://github.com/raspberrypi/pico-sdk/blob/master/CONTRIBUTING.md

MrJake222, 2023-05-07:
Define "against develop". The base for this PR is the master branch, but this PR will merge into develop. Probably this PR needs to be closed because of this.

> undo the effect" of https://github.com/raspberrypi/pico-sdk/pull/1318

This PR is not part of `master` branch (the include isn't there).

drankinatty, 2023-05-14:
How long will the fix take to propagate? As of May 14, 12:39 CST, with a fresh git pull in pico-sdk, the issue is still present.

drankinatty, 2023-05-15:
Thank you. Will pull the development branch.

lurch, 2023-05-22:
IMHO #1367 is a "better" fix than #1364

codydg, 2023-05-23:
> IMHO #1367 is a "better" fix than #1364

Agreed - Either way this issue seems like it should still be open.

kilograham, 2023-05-24:
#1367 now actually merged

PR #1367 (2023-05-24): GCC 13 update and explicit includes. Fixes #1363
GCC 13 now requires explicitly including some libstdc++ headers.

- pioasm: Move `#include <cstdint>` to header from source file. This way it is included in both.

Fixes #1363
commit 7070d230 (2023-05-24): GCC 13 update. Move #include <cstdint> to header. Fixes #1363 (#1367)