raspberrypi/pico-sdk #1371: Add cyw43_arch_disable_sta_mode and cyw43_arch_disable_ap_mode

peterharperuk, 2023-05-11:
It was pointed out that it's a bit inconsistent that these don't exist

kilograham, 2023-05-26:
merged into develop

closed 2023-05-26 by kilograham

rewolff, 2023-10-31:
FYI: It is a bit annoying that the documentation mentions these and then the released version doesn't have them. 

I have a project where I want to join the existing network, or become AP if none is yet available. (several identical boards need to cooperate). This could be quite common, join the configured network, but if it is not available, become AP for configuration. By the "disable" functions being available it is hinted that this would be necessary. (or that simultaneous AP/client mode is possible).

peterharperuk, 2023-10-31:
> FYI: It is a bit annoying that the documentation mentions these and then the released version doesn't have them.

You've lost me, I just checked and these functions are in the master branch. And simultaneous AP/client mode IS possible?
Please open a new issue if you think you have found a problem.

rewolff, 2023-10-31:
Aarg! I didn't do "git pull". I've been checking out an older version for being able to compile an older project. And then switching back to master with git telling me: "Your branch is up to date with 'origin/master'." each time.

peterharperuk, 2023-10-31:
phew

PR #1383 (2023-05-26): Add "disable" functions for STA/AP modes
We have functions to "enable" STA/AP modes, so add versions that do the opposite.

Fixes #1371
commit 5ee4c1c3 (2023-05-26): Add cyw43_arch_disable_sta_mode (#1383)

We have a function to "enable" STA mode, so add a "disable" version to
match AP mode.

Fixes #1371