raspberrypi/pico-sdk #1375: Including External/Custom Libraries

SlazyHW, 2023-05-16:
Hello everyone,

I just started programming my Pico in C/C++ using SDK. I worked through the entire Getting Started guide and I didn't find any clues on how to include custom/external libraries. I know that CMake manages all dependencies but I never worked with it before so it is really confusing for me.

What I want is the following:

I want to add custom libraries somewhere in the pico-sdk folder so I can just add them to my projects CMakeLists.txt by including it in target_link_libraries. When the custom/external libraries depend on default pico-sdk libraries, then these dependencies need to be linked automatically without additional needs.

The libraries I want to use are for example: 

https://github.com/ArmDeveloperEcosystem/lorawan-library-for-pico#readme
https://github.com/ArmDeveloperEcosystem/st7789-library-for-pico/tree/main

How can I achieve this? Thanks in advance!

rspber, 2023-05-16:
The references you point to are not cmake libraries but projects.

peterharperuk, 2023-05-16:
This is a question for the forum really https://forums.raspberrypi.com/viewforum.php?f=145
Might be worth adding feedback on the docs here https://github.com/raspberrypi/pico-feedback/issues

In summary you want to add a lines like this into your project...

include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)
...
pico_sdk_init()

You could physically copy pico_sdk_import.cmake into your folder and include it from there. This will be including the files needed to allow you to link to pico-sdk stuff. Generally you include(somefile.cmake). The first project you referenced doesn't appear to have been built with cmake reuse in mind - you could consider asking for them to add a cmake file for the library you need (or push a change to their repo).

rspber, 2023-05-16:
There is also another method that I use in my repositories, which is that in your project you create a directory that is supposed to represent a foreign library, copy the foreign CMakeList.txt to it and convert it to a library, and to all files add a directory where this file exists.

closed 2023-05-26 by kilograham