raspberrypi/pico-sdk #1378: Need a way to reserve flash sectors

peterharperuk, 2023-05-17:
The BT code uses two sectors at the end of flash for saving pairing info. As more people need to use flash it will get harder to coordinate flash allocations. We need some sort of API that allocates flash memory at runtime to avoid clashes?

peterharperuk, 2023-05-17:
@kilograham

geurtv, 2023-05-31:
Alternatively or additionally maybe provide a more generalized storage callback interface, with some default implementations. This way the user can decide where and how stuff should be stored. For instance if a filesystem is already used (like littlefs or fatfs) the user might want to store configuration values in files. Something like:

```c
struct storage_interface {
    void* user_data;
    size_t (*load_value)(void* udata, const char* key, void* buffer, size_t size);
    void (*store_value)(void* udata, const char* key, const void* buffer, size_t size);
    // ... more? get_value_size?
};
```