raspberrypi/pico-sdk #1381: Define two extra symbols in default linker script for broader compatibility

vkottler, 2023-05-22:
Hello, I've been building the SDK with a [crosstool-ng](https://crosstool-ng.github.io/)-produced toolchain (configuration [here](https://github.com/project-81/firmware/blob/master/toolchains/arm-picolibc-eabi/.config)) that uses [picolibc](https://github.com/picolibc/picolibc) for the C library.

Everything works great, but I did have to add the following to lines to [memmap_default.ld](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_standard_link/memmap_default.ld):

```
PROVIDE (__heap_start = __end__);
PROVIDE (__heap_end = __HeapLimit);
```

[Here](https://github.com/picolibc/picolibc/blob/3915ae4672968d216986b38e99df77a29f16fe18/newlib/libc/picolib/picosbrk.c#L38-L41) is the code that creates the need for these symbols.

If a maintainer thinks this is reasonable to add, I would be happy to submit a pull request.

peterharperuk, 2023-05-22:
We welcome PRs

kilograham, 2024-08-09:
fixed now - we have proper support for picolibc e.g. per LLVM ETA - if you are doing it yourself you may want to pass `PICO_CLIB=picolibc` to CMake

closed 2024-08-09 by kilograham