raspberrypi/pico-sdk #1385: Add ability to save core file to flash

strspn, 2023-05-22:
The SDK contains many points of error checking that result in program termination, but unless the Pico is connected to a host via UART or USB, there is no record left behind. This works well for development, but for deployed products, there is no debug information saved to help solve crashes.

It would be helpful if the SDK could write a core file to the lower portion of flash memory, such that it could be downloaded to a development host system and examined using gdb "target core". This then would also require a way to locate the core file and upload it to a host. Even if only the stack is saved, it would be helpful to determining the cause.

A compile-time define switch passed through cmake would allow users to opt-out to save executable file space. The hooks include *assert*(), panic*(), and __breakpoint().

peterharperuk, 2023-05-22:
Yes, that would be good

strspn, 2023-05-29:
In the case where the full memory image is not saved, the saved stack should include some number of stack bytes/frames above the current stack pointer. This allows the developer to see the frame(s) of call(s) made and returned prior to the crash.