raspberrypi/pico-sdk #1390: Broken on Mac OS X 12.6.6

ricardomatias, 2023-05-23:
I get the following error when cmake tries to configure via VSCode:

```
[main] Configuring project: pirack 
[proc] Executing command: /usr/local/bin/cmake --no-warn-unused-cli -DCMAKE_TOOLCHAIN_FILE:STRING=/Users/feral/vcpkg/scripts/buildsystems/vcpkg.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=/usr/local/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER:FILEPATH=/usr/local/bin/arm-none-eabi-g++ -S/Users/feral/Projects/DIY/pico/pirack -B/Users/feral/Projects/DIY/pico/pirack/build -G Ninja
[cmake] PICO_SDK_PATH is /Users/feral/Projects/DIY/pico/LIBRARIES/pico-sdk
[cmake] Not searching for unused variables given on the command line.
[cmake] PICO platform is rp2040.
[cmake] -- The CXX compiler identification is GNU 10.3.1
[cmake] -- The C compiler identification is GNU 10.3.1
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: /usr/local/bin/arm-none-eabi-gcc
[cmake] -- Checking whether CXX compiler has -isysroot
[cmake] -- Checking whether CXX compiler has -isysroot - yes
[cmake] -- Checking whether CXX compiler supports OSX deployment target flag
[cmake] -- Checking whether CXX compiler supports OSX deployment target flag - no
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - failed
[cmake] -- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++
[cmake] -- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ - broken
[cmake] CMake Error at /usr/local/Cellar/cmake/3.26.4/share/cmake/Modules/CMakeTestCXXCompiler.cmake:60 (message):
[cmake]   The C++ compiler
[cmake] 
[cmake]     "/usr/local/bin/arm-none-eabi-g++"
[cmake] 
[cmake]   is not able to compile a simple test program.
[cmake] 
[cmake]   It fails with the following output:
[cmake] 
[cmake]     Change Dir: /Users/feral/Projects/DIY/pico/pirack/build/CMakeFiles/CMakeScratch/TryCompile-MEMGpa
[cmake]     
[cmake]     Run Build Command(s):/usr/local/bin/ninja -v cmTC_90f11 && [1/2] /usr/local/bin/arm-none-eabi-g++   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -o CMakeFiles/cmTC_90f11.dir/testCXXCompiler.cxx.o -c /Users/feral/Projects/DIY/pico/pirack/build/CMakeFiles/CMakeScratch/TryCompile-MEMGpa/testCXXCompiler.cxx
[cmake]     [2/2] : && /usr/local/bin/arm-none-eabi-g++ -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names -L/usr/local/opt/opencv@2/lib CMakeFiles/cmTC_90f11.dir/testCXXCompiler.cxx.o -o cmTC_90f11   && :
[cmake]     FAILED: cmTC_90f11 
[cmake]     : && /usr/local/bin/arm-none-eabi-g++ -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names -L/usr/local/opt/opencv@2/lib CMakeFiles/cmTC_90f11.dir/testCXXCompiler.cxx.o -o cmTC_90f11   && :
[cmake]     /usr/local/Cellar/arm-none-eabi-gcc/10.3-2021.07/gcc/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: Error: unable to disambiguate: -search_paths_first (did you mean --search_paths_first ?)
[cmake]     collect2: error: ld returned 1 exit status
[cmake]     ninja: build stopped: subcommand failed.
[cmake]     
[cmake]     
[cmake] 
[cmake]   
[cmake] 
[cmake]   CMake will not be able to correctly generate this project.
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:9 (project)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[proc] The command: /usr/local/bin/cmake --no-warn-unused-cli -DCMAKE_TOOLCHAIN_FILE:STRING=/Users/feral/vcpkg/scripts/buildsystems/vcpkg.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=/usr/local/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER:FILEPATH=/usr/local/bin/arm-none-eabi-g++ -S/Users/feral/Projects/DIY/pico/pirack -B/Users/feral/Projects/DIY/pico/pirack/build -G Ninja exited with code: 1
```

lurch, 2023-05-23:
Is this on an Intel Mac or a Apple Silicon Mac? Have you followed the instructions in section 9.1 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ?

ricardomatias, 2023-05-23:
Intel Mac, I just upgraded to Ventura 13.4 and the issue persists. Yes, I've got `arm-none-eabi-gcc` installed via homebrew.

kilograham, 2023-05-23:
first thing, try deleteing your build directory if you have upgraded macOS

kilograham, 2023-05-23:
also `-DCMAKE_TOOLCHAIN_FILE:STRING=/Users/feral/vcpkg/scripts/buildsystems/vcpkg.cmake` looks suspicious

anyways, better discussed on the forums https://forums.raspberrypi.com/viewforum.php?f=143