raspberrypi/pico-sdk #1392: Communication with cyw43 fails at system clock speeds <= 60MHz

MichaelBell, 2023-05-25:
It's sometimes useful to run the RP2040 quite slowly to reduce power consumption.  I'm not sure whether it is reasonable to expect Wifi to continue to work at low clock speeds - but I think it is reasonable to at least expect to be able to use the CYW43 GPIO.

However, at around 60MHz and below setting/reading the GPIO from the CYW43 doesn't work.  This means you can't determine whether VBUS is present, and can't control the LED.

A simple repro is to add set_sys_clock_48mhz() to the top of the pico_w blink example.

```diff
index 149b477..600e2a2 100644
--- a/pico_w/wifi/blink/picow_blink.c
+++ b/pico_w/wifi/blink/picow_blink.c
@@ -8,6 +8,7 @@
 #include "pico/cyw43_arch.h"

 int main() {
+    set_sys_clock_48mhz();
     stdio_init_all();
     if (cyw43_arch_init()) {
         printf("Wi-Fi init failed");
```

thingsconnected, 2023-10-23:
WiFi works flawlessly for us with the Pi Pico running at 12 MHz from xosc. Set clock divider at 1 and spi program spi_gap01_sample0. Running at 48 MHz (to enable USB stdio) clock divider 2 and spi program spi_gap0_sample1 can be used, IIRC.

peterharperuk, 2024-08-09:
I seem to be able to be able to use cyw43 to flash the led when running from xosc by calling the new function cyw43_set_pio_clock_divisor(1, 0). I didn't seem to have to set CYW43_SPI_PROGRAM_NAME? Anyway - I think this is resolved now as you shouldn't have to hack the code?

closed 2024-08-09 by peterharperuk