raspberrypi/pico-sdk #1396: Cannot compile test executable because cmake tries to run "PioasmBuild" and "ELF2UF2Build" on this

ricardomatias, 2023-05-26:
As it shows at the end of the stack trace, it tries to link against "sonovolt_test.elf" which shouldn't exist. The test is inside a local library that uses the pico std.

CMakeLists.txt:

```
add_library( ${LIB_NAME} STATIC)

target_compile_features(${LIB_NAME} PUBLIC cxx_std_17)

target_sources(${LIB_NAME}
    PRIVATE
        ${CMAKE_CURRENT_LIST_DIR}/src/common/math.cpp
        ${CMAKE_CURRENT_LIST_DIR}/src/common/time.cpp
        ${CMAKE_CURRENT_LIST_DIR}/src/pico/pico.cpp
)

target_include_directories(${LIB_NAME} PUBLIC ${CMAKE_CURRENT_LIST_DIR}/include)

target_link_libraries(${LIB_NAME} 
    PUBLIC 
        hardware_pwm
)

set(TEST_EXE ${LIB_NAME}_test)

add_executable(${TEST_EXE} ${CMAKE_CURRENT_LIST_DIR}/test/time.cpp)

target_compile_features(${TEST_EXE} PRIVATE cxx_std_17)

find_path(DOCTEST_INCLUDE_DIR NAMES doctest.h PATH_SUFFIXES doctest)
message(STATUS "doctest include path: ${DOCTEST_INCLUDE_DIR}")
target_include_directories(${TEST_EXE} 
    PRIVATE 
        ${DOCTEST_INCLUDE_DIR}
        ${CMAKE_CURRENT_LIST_DIR}/include
)
```

```
[build] [ 35%] [32m[1mLinking CXX executable pirack.elf[0m
[build] [ 98%] Built target pirack
[build] In file included from c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\vector:70,
[build]                  from C:\Users\Ricardo\code\pico\pirack\build\vcpkg_installed\arm-mingw-static\include\doctest/doctest.h:3169,
[build]                  from C:\Users\Ricardo\code\pico\pirack\extern\sonovolt\test\time.cpp:7:
[build] c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\bits\vector.tcc: In member function 'void std::vector<_Tp, _Alloc>::_M_realloc_insert(iterator, _Args&& ...) [with _Args = {const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}; _Tp = doctest::{anonymous}::JUnitReporter::JUnitTestCaseData::JUnitTestCase; _Alloc = std::allocator<doctest::{anonymous}::JUnitReporter::JUnitTestCaseData::JUnitTestCase>]':
[build] c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\bits\vector.tcc:439:7: note: parameter passing for argument of type 'std::vector<doctest::{anonymous}::JUnitReporter::JUnitTestCaseData::JUnitTestCase>::iterator' changed in GCC 7.1
[build]   439 |       vector<_Tp, _Alloc>::
[build]       |       ^~~~~~~~~~~~~~~~~~~
[build] c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\bits\vector.tcc: In member function 'std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::emplace_back(_Args&& ...) [with _Args = {const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&}; _Tp = doctest::{anonymous}::JUnitReporter::JUnitTestCaseData::JUnitTestCase; _Alloc = std::allocator<doctest::{anonymous}::JUnitReporter::JUnitTestCaseData::JUnitTestCase>]':
[build] c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\bits\vector.tcc:123:28: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<doctest::{anonymous}::JUnitReporter::JUnitTestCaseData::JUnitTestCase*, std::vector<doctest::{anonymous}::JUnitReporter::JUnitTestCaseData::JUnitTestCase> >' changed in GCC 7.1
[build]   123 |           _M_realloc_insert(end(), std::forward<_Args>(__args)...);
[build]       |           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\bits\vector.tcc: In member function 'void std::vector<_Tp, _Alloc>::_M_realloc_insert(iterator, _Args&& ...) [with _Args = {std::fpos<_mbstate_t>}; _Tp = std::fpos<_mbstate_t>; _Alloc = std::allocator<std::fpos<_mbstate_t> >]':
[build] c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\bits\vector.tcc:439:7: note: parameter passing for argument of type 'std::vector<std::fpos<_mbstate_t> >::iterator' changed in GCC 7.1
[build]   439 |       vector<_Tp, _Alloc>::
[build]       |       ^~~~~~~~~~~~~~~~~~~
[build] c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\bits\vector.tcc: In member function 'std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::emplace_back(_Args&& ...) [with _Args = {std::fpos<_mbstate_t>}; _Tp = std::fpos<_mbstate_t>; _Alloc = std::allocator<std::fpos<_mbstate_t> >]':
[build] c:\users\ricardo\scoop\apps\gcc-arm-none-eabi\current\arm-none-eabi\include\c++\12.2.1\bits\vector.tcc:123:28: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<std::fpos<_mbstate_t>*, std::vector<std::fpos<_mbstate_t> > >' changed in GCC 7.1
[build]   123 |           _M_realloc_insert(end(), std::forward<_Args>(__args)...);
[build]       |           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] [100%] [32m[1mLinking CXX executable sonovolt_test.elf[0m
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-exit.o): in function `exit':
[build] exit.c:(.text.exit+0x16): undefined reference to `_exit'
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-abort.o): in function `abort':
[build] abort.c:(.text.abort+0xa): undefined reference to `_exit'
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-closer.o): in function `_close_r':
[build] closer.c:(.text._close_r+0xc): undefined reference to `_close'
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-fstatr.o): in function `_fstat_r':
[build] fstatr.c:(.text._fstat_r+0xe): undefined reference to `_fstat'
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-gettimeofdayr.o): in function `_gettimeofday_r':
[... 1 lines cut ...]
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-lseekr.o): in function `_lseek_r':
[build] lseekr.c:(.text._lseek_r+0x10): undefined reference to `_lseek'
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-openr.o): in function `_open_r':
[build] openr.c:(.text._open_r+0x10): undefined reference to `_open'
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-readr.o): in function `_read_r':
[build] readr.c:(.text._read_r+0x10): undefined reference to `_read'
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp\libg.a(libc_a-signalr.o): in function `_kill_r':
[build] signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
[build] c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/ricardo/scoop/apps/gcc-arm-none-eabi/current/bin/../lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m [... cut ...]
[build] mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:1734: extern/sonovolt/CMakeFiles/sonovolt_test.dir/all] Error 2
[build] mingw32-make.exe: *** [Makefile:155: all] Error 2
[proc] The command: "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.EXE" --build c:/Users/Ricardo/code/pico/pirack/build --config Debug --target all -j 14 -- exited with code: 2
[driver] Build completed: 00:00:03.311
[build] Build finished with exit code 2

```

peterharperuk, 2023-05-26:
I don't understand this issue. Can you put together an example that demonstrates the problem you're reporting?

ricardomatias, 2023-05-29:
Closing this since the issue was on my side.

closed 2023-05-29 by ricardomatias