raspberrypi/pico-sdk #1432: CMake issue

pierrotm777, 2023-06-18:
Hello,

I try to install the new version 1.5.1 after my old 1.5.0 uninstalled.
I obtain always this error:
`Unable to resolve configuration with compilerPath: "c:/program files/raspberry pi/pico sdk v1.5.0/gcc-arm-none-eabi/bin/arm-none-eabi-g++.exe"`

What i can do for the compiler use the new path ?
Thanks for your help,
Pierre

kilograham, 2023-06-18:
Question best asked on the forums https://forums.raspberrypi.com/viewforum.php?f=143

but these lines

```
[cmake] Using PICO_SDK_PATH from environment ('..\..\pico-sdk')
[cmake] CMake Error at pico_sdk_import.cmake:63 (message):
[cmake]   Directory 'C:/Users/pierrot/Documents/Pico-v1.5.1/pico-sdk/pico-sdk' not
[cmake]   found
```

would definitely seem to suggest your PICO_SDK_PATH Is set incorrectly; better to use an absolute path.

closed 2023-06-18 by kilograham

pierrotm777, 2023-06-19:
Ok, i understand but i don't know witch command i need to do.
My Pico SDK is in this path:
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1
My projects folder is in the same path that the examples files:
C:\Users\pierrot\Documents\Pico-v1.5.1

I have tried to add PICO_SDK_PATH and ..\..\pico-sdk into my PATH system without success.
I have also tried to add this same path into Vscode in the CMake Environnement settings without success.

What good solution the the best ?

Thanks,

Pierre

kilograham, 2023-06-19:
yes, please ask on the forums mentioned above

lurch, 2023-06-21:
You might also want to ask at https://github.com/raspberrypi/pico-setup-windows/issues