raspberrypi/pico-sdk #1434: PICO_ASSEMBLER_IS_CLANG failing on Arch

bill88t, 2023-06-19:
Hello, first things first, systems it happens on:
```
RebornOS Linux ARM aarch64 Raspberry Pi 400 Rev 1.0 6.1.34-1-rpi-ARCH

aur/gcc-arm-none-eabi-bin 12.2.rel1-1 (+27 0.01) (Installed)
    GNU Tools ARM Embedded Processors (binary distribution, includes newlib, does NOT include GDB)
```
```
Arch Linux x86_64 MS-7C83 1.0 6.3.8-arch1-1

extra/arm-none-eabi-binutils 2.40-1 (3.6 MiB 18.8 MiB) (Installed)
    A set of programs to assemble and manipulate binary and object files for the ARM EABI (bare-metal) target
extra/arm-none-eabi-newlib 4.3.0.20230120-1 (17.6 MiB 242.2 MiB) (Installed)
    A C standard library implementation intended for use on embedded systems (ARM bare metal)
extra/arm-none-eabi-gcc 13.1.0-1 (218.9 MiB 1.7 GiB) (Installed)
    The GNU Compiler Collection - cross compiler for ARM EABI (bare-metal) target
extra/arm-none-eabi-gdb 13.1-2 (3.3 MiB 8.7 MiB) (Installed)
    The GNU Debugger for the ARM EABI (bare-metal) target
```

And the error is as follows (this was captured on the x86 machine):
```
PICO_SDK_PATH is /home/bill88t/git/circuitpython/ports/raspberrypi/sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/bill88t/git/circuitpython/ports/raspberrypi/sdk/src/boards/include/boards/pico.h
GEN build-raspberry_pi_pico_w/genhdr/moduledefs.h
AS supervisor/rp2_cpu.s
CC sdk/src/rp2_common/hardware_divider/divider.S
CC sdk/src/rp2_common/hardware_irq/irq_handler_chain.S
CC sdk/src/rp2_common/pico_bit_ops/bit_ops_aeabi.S
CC sdk/src/rp2_common/pico_double/double_aeabi.S
CC sdk/src/rp2_common/pico_double/double_v1_rom_shim.S
CC sdk/src/rp2_common/pico_float/float_aeabi.S
CC sdk/src/rp2_common/pico_divider/divider.S
CC sdk/src/rp2_common/pico_float/float_v1_rom_shim.S
CC sdk/src/rp2_common/pico_int64_ops/pico_int64_ops_aeabi.S
CC sdk/src/rp2_common/pico_mem_ops/mem_ops_aeabi.S
CC sdk/src/rp2_common/pico_standard_link/crt0.S
CC lib/PicoDVI/software/libdvi/tmds_encode_asm.S
CC build-raspberry_pi_pico_w/x509_crt_bundle.S
sdk/src/rp2_common/pico_standard_link/crt0.S:336:5: error: "PICO_ASSEMBLER_IS_CLANG" is not defined, evaluates to 0 [-Werror=undef]
  336 | #if PICO_ASSEMBLER_IS_CLANG
      |     ^~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:28: build-raspberry_pi_pico_w/sdk/src/rp2_common/pico_standard_link/crt0.o] Error 1
make: *** Waiting for unfinished jobs....
```

Clang is also installed.
What changes do I need to perform to the build env?

kilograham, 2023-06-19:
You must have enabled `-Wundef` and `-Werror` somewhere in your build (`CMakeLists.txt`)... remove the `-Wundef`

is this a regression with 1.5.1?

bill88t, 2023-06-19:
I was trying to update pico-sdk on circuitpython to use some of the new cyw43 funcs.
The previous version of pico-sdk CP was on was 1.5.0, commit 2ccab11.

CircuitPython indeed has a `-Wundef` in place (very high up, `py/circuitpy_defns.mk`), and removing it, fixes this issue.

So yea, it's regression time.

Should I leave this issue open?
Is there some action that needs to be taken, or should I close it?

bill88t, 2023-06-19:
Actually, an `#ifdef` is all it should take, so yea.
I don't like how it looks, but it's 1 am.

kilograham, 2024-05-19:
we don't plan to support `-Wundef`, so closing

closed 2024-05-19 by kilograham