raspberrypi/pico-sdk #1440: C++ Overriding New and Delete

jondurrant, 2023-06-30:
New and Delete can only be overridden when C++ Exceptions are enabled with set(PICO_CXX_ENABLE_EXCEPTIONS 1).  Yet there are times when  I may wish to override New and Delete operators without using Exceptions.

My context issuing FreeRTOS Heap model for New and Delete.

geurtv, 2023-07-01:
There's PICO_CXX_DISABLE_ALLOCATION_OVERRIDES since I guess pico-sdk 1.5.0, so with `add_compile_definitions(PICO_CXX_DISABLE_ALLOCATION_OVERRIDES=1)` pico-sdk won't override operators new and delete anymore.

jondurrant, 2023-07-01:
Thank you.  Missed that one.

Jon

kilograham, 2024-05-19:
There is a solution here, but i'm keeping this around to remind me as I look at cleaning up the interface with the C/C++ library

kilograham, 2024-07-21:
i think the solution is fine for now, so closing

closed 2024-07-21 by kilograham