raspberrypi/pico-sdk #1442: Please allow custom .gatt #import paths in pico_btstack_make_gatt_header()

rppicomidi, 2023-06-30:
I am trying to develop a custom Bluetooth LE GATT service for the Pico W that can be imported into another `.gatt` file using `#import <my_custom.gatt>`. I don't want to try to add this service to the BlueKitchen Bluetooth stack's existing default paths for `.gatt` files.

The file `${PICO_SDK_PATH}/src/rp2_common/pico_btstack/CMakeLists.txt` contains the function `pico_btstack_make_gatt_header()` that is usually invoked like this:
```
pico_btstack_make_gatt_header(my_executable PRIVATE "${CMAKE_CURRENT_LIST_DIR}/my_gap_device.gatt")
```
This causes the Python script `${PICO_SDK_PATH}/lib/btstack/tool/compile_gatt.py` to be invoked with a path to the `.gatt` file and the path to the header file. I need to the ability to add `-I${MY_CUSTOM_GATT_FILE_PATH}` to the end of the argument list so the tool can find my custom `.gatt` file when it reaches the `#import` line that calls for it. That is, I want my project's CMakeLists.txt to be able to call
```
pico_btstack_make_gatt_header(my_executable PRIVATE "${CMAKE_CURRENT_LIST_DIR}/my_gap_device.gatt" -I${MY_CUSTOM_GATT_FILE_PATH})
```

I have a simple fix for this on my local machine and I would like to submit it as a pull request. Should I submit my change as a fork from `master` or `develop` branch?

peterharperuk, 2023-06-30:
PR's against develop please

rppicomidi, 2023-07-01:
Submitted #1445

kilograham, 2024-01-16:
merged into devleop

commit f1c6fc8b (2024-01-16): Fix #1442: Allow adding alternative .gatt file import paths  (#1445)

* fix feature request #1442

* Do not require -I before each addtional path

* Fix pico_btstack_make_gatt_header warning

gatt header files are always made into the "generated" folder so you get
a warning if you have more than one target generating a gatt header with
the same name.

Also, simplify the expansion of ARGN

* Improve pico_btstack_make_gatt_header description

---------
closed 2024-01-16 by kilograham