raspberrypi/pico-sdk #1443: Documentation for gpio_add_raw_irq_handler functions

hubiscode, 2023-06-30:
The documentation for the gpio_add_raw_irq_handler*() functions does not mention that there is a limitation on the number of handlers that can be added. The limitation is due to the call to irq_add_shared_handler(), which has a limit that can be configured via PICO_MAX_SHARED_IRQ_HANDLERS. The default value in the SDK is a maximum of four handlers, of which two or three may be used by USB and the cyw43 driver. I think it would be helpful to mention the limitation since anybody installing more than one or two handlers is likely to exceed the default limit.

MichaelBell, 2023-09-19:
Oh, I didn't realise the raw handler functions were using shared handler underneath.  That would probably explain why I'm hitting a problem, as it is extremely easy to run out of shared handlers!

kilograham, 2024-05-19:
merged into develop

PR #1444 (2024-01-13): Fix spelling mistake, add documentation
Add documentation to gpio_add_raw_irq_handler*() functions that references irq_add_shared_handler() and notes that the functions will assert if the maximum number of shared handlers configured via PICO_MAX_SHARED_IRQ_HANDLERS would be exceeded.

Fix spelling mistake in static variable in irq.c (hander vs handler).

Fixes #1443
closed 2024-05-19 by kilograham