raspberrypi/pico-sdk #1449: error: redeclaration of enumerator 'HID_REPORT_TYPE_INPUT'

rynhardthaarhoff, 2023-07-03:
When including both **tinyusb** and **btstack** libraries, I get the following error:
```
In file included from C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/btstack/src/btstack.h:62,
[build]                  from C:/Users/Username/Documents/Pico-v1.5.1/pico-projects/Solo/main.c:32:
[build] C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/btstack/src/btstack_hid.h:103:5: error: redeclaration of enumerator 'HID_REPORT_TYPE_INPUT'
[build]   103 |     HID_REPORT_TYPE_INPUT,
[build]       |     ^~~~~~~~~~~~~~~~~~~~~
[build] In file included from C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/tinyusb/src/tusb.h:41,
[build]                  from C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/tinyusb/hw/bsp/board.h:42,
[build]                  from C:/Users/Username/Documents/Pico-v1.5.1/pico-projects/Solo/main.c:28:
[build] C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/tinyusb/src/class/hid/hid.h:87:3: note: previous definition of 'HID_REPORT_TYPE_INPUT' was here
[build]    87 |   HID_REPORT_TYPE_INPUT,      ///< Input
```

This affects the following enums: 
- HID_REPORT_TYPE_INPUT
- HID_REPORT_TYPE_OUTPUT 
- HID_REPORT_TYPE_FEATURE

I have resolved the issue by renaming these enums in the btstack library, in the files 
- btstack_hid.h
- btstack_hid_parser.c
- btstack_hid_parser.h
- hids_client.c
- hids_client.h

It may be that I'm not doing the linking correctly. Here is my CMakelists.txt file:
```
cmake_minimum_required(VERSION 3.13)
set(PICO_BOARD pico_w)
include(pico_sdk_import.cmake)
project(HoffSolo C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
pico_sdk_init()

add_compile_options(-Wall
        -Wno-format          # int != int32_t as far as the compiler is concerned because gcc has int32_t as long int
        -Wno-unused-function # we have some for the docs that aren't called
        )
if (CMAKE_C_COMPILER_ID STREQUAL "GNU")
    add_compile_options(-Wno-maybe-uninitialized)
endif()

add_subdirectory(lib/screen/config)
add_subdirectory(lib/screen/lcd)
add_subdirectory(lib/screen/font)

include_directories(./lib/screen/config)
include_directories(./lib/screen/lcd)
include_directories(./lib/screen/font)
include_directories(./lib/tusb)
include_directories(./lib/hoff)
include_directories(./lib/ble)

add_executable(solo)
target_sources(solo PRIVATE
        ${CMAKE_CURRENT_LIST_DIR}/main.c
        ${CMAKE_CURRENT_LIST_DIR}/lib/tusb/usb_descriptors.c
        )
target_link_libraries(solo PRIVATE
        pico_stdlib
        pico_btstack_cyw43
        pico_cyw43_arch_threadsafe_background
        pico_btstack_ble
        tinyusb_device
        tinyusb_board
        lcd
        font
        config
        pico_stdlib
        hardware_spi
        hardware_flash
        pico_multicore
        )
target_include_directories(solo PRIVATE
        ${CMAKE_CURRENT_LIST_DIR}
        ${PICO_SDK_PATH}/lib/btstack/src
        )
target_compile_definitions(solo PRIVATE
        PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS=3000
        CYW43_LWIP=0
        )
pico_add_extra_outputs(solo)

pico_btstack_make_gatt_header(solo PRIVATE ${CMAKE_CURRENT_LIST_DIR}/lib/ble/solo_midi_ble.gatt)
```

peterharperuk, 2023-07-03:
That's annoying!

WayneKeenan, 2023-07-23:
@rynhardthaarhoff  If you separate your btstack and tinyusb C code to separate compilation units it is not annoying anymore.

peterharperuk, 2023-10-09:
As @WayneKeenan says - you can try to avoid this by not mixing BT and USB code.

WayneKeenan, 2023-10-09:
Also, as these are upstream/3rd party libraries there's not really a suitable 'fix' that lies within in the scope of the Pico SDK.

thewh1teagle, 2023-12-29:
@peterharperuk 
What if I use them both in my code?
How can I compile it?

peterharperuk, 2023-12-29:
It's a simple matter to not use usb and bt headers in the same source file.

closed 2023-12-29 by peterharperuk