raspberrypi/pico-sdk #1450: set_sys_clock_khz allows only Mhz

webdeb, 2023-07-04:
Although the function name is `set_sys_clock_khz` the values have to be fixed Mhz like 100000 khz, 101000 khz etc.
What is the reason for this limitation?

lurch, 2023-07-05:
https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#gab3a273e837ba1947bb5fd8fc97cf47e5 says "Note that not all clock frequencies are possible; it is preferred that you use src/rp2_common/hardware_clocks/scripts/vcocalc.py to calculate the parameters for use with set_sys_clock_pll".
You probably want to have a read of section 2.15 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf for more information about the PLLs and clock dividers.

webdeb, 2023-07-05:
thx @lurch I've read the comment, before posting the issue. However, I tried all 1000 khz values between 125.001 and 126 MHz. Only full Mhz worked. I'll definitely check out the datasheet sections you mentioned. However, do you mind telling me, whats the most quick way to change the freq in the range of 100x Mhz with max precision?

kilograham, 2023-07-05:
closing this, as you absolutely can do non Mhz frequencies (just not those between 125 and 126)

closed 2023-07-05 by kilograham

webdeb, 2023-07-05:
Ah, ok, the idea was to change the clock as smooth as possible. But it will not work, I see. Nevertheless thank you.