raspberrypi/pico-sdk #1451: vcocalc.py should have an option to use the default REFDIV of 2

kilograham, 2023-07-05:
#952 added support for non-standard REFDIV values in vcocalc.py, however the default now avoids the default REFDIV value used by the code (2).

This is a bit mean (and there is no way to set the minimum and maximum such that only 2 is specified)

kilograham, 2023-07-05:
cc @alastairpatrick

Note, REFDIV is only settable by `#define` which was intentional as it is common, perhaps we should add some blurb to the vcocalc.py output to say how to set up that clock

kilograham, 2023-07-05:
note actually, maybe we should default both refdivs min/max to 2, or default them to 1 and 63 as they are now, but add an flag to allow you to pick a REFDIV other than 2

alastairpatrick, 2023-07-27:
Regarding "REFDIV is only settable by #define which was intentional as it is common," is that actually true? I haven't tried but the RP2040 datasheet suggests that the SYS and USB PLLs have independent REFDIVs, so if that's actually true, it might make sense to have separate REFDIV #defines for each PLL.

Alternatively, the USB PLL will basically always want to run at 48MHz. 48MHz can be achieved with both REFDIV=1 and REFDIV=2:
```
Requested: 48.0 MHz
Achieved: 48.0 MHz
REFDIV: 1
FBDIV: 120 (VCO = 1440.0 MHz)
PD1: 6
PD2: 5
```
So it should be possible to make things work with just a single common REFDIV #define, set to either REFDIV=1 or REFDIV=2, have USB clock domain always operate at 48MHz and the SYS clock domain operate at whatever the desired frequency was passed to vcocalc.py.

lurch, 2023-07-27:
> it might make sense to have separate REFDIV #defines for each PLL.

This was (kind of) touched upon recently in #1272

will-v-pi, 2024-08-09:
Fixed with SDK 2.0.0

closed 2024-08-09 by will-v-pi