raspberrypi/pico-sdk #1452: Malfunction at restart

Lizethgm, 2023-07-06:
Hi, 
I have an error in the program generated from pico-sdk 

If I load to my board which has a RP2040 chip a .uf2 generated from the pico-sdk it works correctly when loading the first time, but once the board is turned off and on again the program is still there but does not initialize. Sometimes it works when restart but it is very rare when it does. The error can be seen in the following video.
https://github.com/raspberrypi/pico-sdk/assets/105874681/0ffbdf30-0f8f-4cbd-be42-b2b82784c5af

If I use the core arduino-pico (https://github.com/earlephilhower/arduino-pico), it works correctly, it doesn't have any problem, it initializes all the time even if it is turned off and on again. This can be seen in the following video 
https://github.com/raspberrypi/pico-sdk/assets/105874681/6911327e-8357-4952-a037-0923869e32fd

Do you have any idea what could be happening or what is missing to add to the pico-sdk?

lurch, 2023-07-06:
It looks like you're using a 3rd-party RP2040 board, rather than a Raspberry Pi Pico? Sometimes they use Flash chips which need different settings than the Flash chips used on the official Pico boards. See e.g. #1304 and #401

lurch, 2023-07-10:
Perhaps you'd like to encourage the maker of your particular RP2040 board to submit a board-header, to make things easier for future users? https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards

kilograham, 2024-05-19:
closing as I don't see an issue here

closed 2024-05-19 by kilograham