raspberrypi/pico-sdk #1465: bluetooth/cyw43439: LE Data Length Extensions (DLE) not working properly

mringwal, 2023-07-29:
I'm filing an issue against the CYW43439 reported by a user on the BTstack mailing list.

When ENABLE_LE_DATA_LENGTH_EXTENSION is added to the btstack_config.h, BTstack will enable DLE by sending a LE Write Suggested Default Data Length with 251 bytes and 2120 us.

In the test, the user connects to the Pico W from an iPhone. After connection setup, the data length is exchanged and we receive this event:

Jan 01 18:00:14.214  HCI Event        LE - Data Length Change  RECV  
	Parameter Length: 11 (0x0B)
	Connection Handle: 0x0040
	Max Tx Octets: 251 (0xFB)
	Max Tx Time: 328 (0x148)
	Max Rx Octets: 251 (0xFB)
	Max Rx Time: 2120 (0x848)

Weirdly, while Tx octets is set to the expected 251, the Tx time is only 328 us which does only allow for the minimal 27 bytes packets.

The air trace from a Nordic sniffer shows that the 43439 sends a LL_LENGTH_RSP with the values shown above and later that only small packets are sent.

If the CYW43449 respects the 251 octets/328 us, it's clear that it cannot send larger packets, but we're wondering why it sets Tx timer to 328 us instead of the expected 2120 us

The HCI log as well as the air traces are available here: https://drive.google.com/file/d/1ETqv2eZ10VQFjd5k935EAb6Hg-f1E4D4/view?usp=sharing

Sending large 251 bytes packet instead of the smaller 27 bytes packets roughly doubles throughput, or not.

@peterharperuk could you ask Infineon about this?

jchatterton443, 2023-07-29:
Thanks for opening this for me Matthias;  I'm new to Github/opening issues.  I also uploaded the Ellisys trace [here](https://drive.google.com/file/d/1wt8i-WJQjEYh1XQccrsbLdvACWLB-WzC/view?usp=sharing)

Thanks for everyone's help.

peterharperuk, 2023-09-06:
@mringwal Is this related to https://github.com/raspberrypi/pico-sdk/issues/1342 "BLE Host is only able to receive one packet per interval". I think you'd done some analysis of that and found that there was a limit of 27 bytes?

mringwal, 2023-09-06:
@peterharperuk At the moment, I don't think that these two issues are related. Event with 27 byte PDUs, there should be multiple PDUs per connection interval in #1342

peterharperuk, 2023-09-28:
We have some new firmware that may fix this problem. The firmware is stored in pic-sdk/lib/cyw43-driver. After updating it should show the version...

BT FW download, version = CYW4343A2_001.003.016.**0065**.0000_Generic_SDIO_37MHz_wlbga_BU_RPI_dl_signed

See https://github.com/peterharperuk/cyw43-driver/tree/firmware_dle_fix or use the attached patch file. 

[0001-Update-firmware-for-Data-Length-Extension-fix.patch.zip](https://github.com/raspberrypi/pico-sdk/files/12750460/0001-Update-firmware-for-Data-Length-Extension-fix.patch.zip)

cd pico-sdk/lib/cyw43-driver
git am 0001-Update-firmware-for-Data-Length-Extension-fix.patch

It would be great if someone could test this, otherwise I'm trying to get hold of a sniffer, but it'll take a few days before I can test it.

peterharperuk, 2023-10-02:
I gt my sniffer working and my interpretation of the log is that the firmware fix seemed to work?
[before-after-firmware-change.zip](https://github.com/raspberrypi/pico-sdk/files/12784927/before-after-firmware-change.zip)

mringwal, 2023-10-03:
Hi @peterharperuk Yes, after applying this patch the DLE setup looks correct and I also get a faster throughput to my iPhone. That's a clear improvement.

Slion, 2025-03-27:
On SDK v2.1.1 even with `ENABLE_LE_DATA_LENGTH_EXTENSION` notify only works with up to 20 bytes.

matsobdev, 2025-03-27:
`CYW43439~ driver was updated to 1.1.0 version and it includes firmware fix for DLE. SDK 2.1.1 was released before that.

kilograham, 2025-03-27:
Seems like a 2.1.2 thing if we just need to bump `cyw43_driver`?

peterharperuk, 2025-03-27:
Yes, It's been so long I'd forgotten about it. @Slion could you pull the latest version of lib/cyw43_driver and check it fixes your problem?

mringwal, 2025-03-28:
Side note: DLE and the ATT MTU are kind of orthogonal: you can have an ATT MTU of 155 event with regular 27 bytes ACL packets (the Controller will send multiple of them to send a single large ACL packet) and even with DLE enabled, and a max ACL packet size of 251. the ATT MTU could still be stuck at 23. 

If it doesn't work after updating the cyw43_driver, please send HCI Packet Log, too.

Slion, 2025-03-30:
@peterharperuk Really busy putting together a demo at the moment. I won't have time to test this soon. We worked around it by doing a characteristic read after receiving the notification. For us even 251 bytes might not be enough anyway.

peterharperuk, 2025-05-29:
Marking as fixed because the DLE change is in develop

closed 2025-05-29 by peterharperuk