raspberrypi/pico-sdk #1467: Compiling application with CMAKE_BUILD_TYPE=Release: undefined symbol: stdout

rgrr, 2023-08-02:
Hello,

this is perhaps related to #196...

Currently I'm trying to compile my project with clang (main branch of llvm, embedded toolchain).

`-DCMAKE_BUILD_TYPE=Debug` works, best: the generated image works.

But... compiling with `-DCMAKE_BUILD_TYPE=Release` produces a linker error:

```
[228/228] : && /home/hardy/bin/llvm-arm-none-eabi/bin/clang --target=armv6m-none-eabi -mfloat-abi=soft -march=armv6m --sysroot /home/hardy/bin/llvm-arm-none-eabi/lib/clang-runtimes/arm-none-eabi/armv6m_soft_nofp -O3 -DNDEBUG -Wl,--build-id=none -Xlinker --wrap=sprintf -Xlinker --wrap=snprintf -Xlinker --wrap=vsnprintf -Xlinker --wrap=__clzsi2 -Xlinker --wrap=__clzdi2 -Xlinker --wrap=__ctzsi2 -Xlinker --wrap=__ctzdi2 -Xlinker --wrap=__popcountsi2 -Xlinker --wrap=__popcountdi2 -Xlinker --wrap=__clz -Xlinker --wrap=__clzl -Xlinker --wrap=__clzll -Xlinker --wrap=__aeabi_idiv -Xlinker --wrap=__aeabi_idivmod -Xlinker --wrap=__aeabi_ldivmod -Xlinker --wrap=__aeabi_uidiv -Xlinker --wrap=__aeabi_uidivmod -Xlinker --wrap=__aeabi_uldivmod -Xlinker --wrap=__aeabi_dadd -Xlinker --wrap=__aeabi_ddiv -Xlinker --wrap=__aeabi_dmul -Xlinker --wrap=__aeabi_drsub -Xlinker --wrap=__aeabi_dsub -Xlinker --wrap=__aeabi_cdcmpeq -Xlinker --wrap=__aeabi_cdrcmple -Xlinker --wrap=__aeabi_cdcmple -Xlinker --wrap=__aeabi_dcmpeq -Xlinker --wrap=__aeabi_dcmplt -Xlinker --wrap=__aeabi_dcmple -Xlinker --wrap=__aeabi_dcmpge -Xlinker --wrap=__aeabi_dcmpgt -Xlinker --wrap=__aeabi_dcmpun -Xlinker --wrap=__aeabi_i2d -Xlinker --wrap=__aeabi_l2d -Xlinker --wrap=__aeabi_ui2d -Xlinker --wrap=__aeabi_ul2d -Xlinker --wrap=__aeabi_d2iz -Xlinker --wrap=__aeabi_d2lz -Xlinker --wrap=__aeabi_d2uiz -Xlinker --wrap=__aeabi_d2ulz -Xlinker --wrap=__aeabi_d2f -Xlinker --wrap=sqrt -Xlinker --wrap=cos -Xlinker --wrap=sin -Xlinker --wrap=tan -Xlinker --wrap=atan2 -Xlinker --wrap=exp -Xlinker --wrap=log -Xlinker --wrap=ldexp -Xlinker --wrap=copysign -Xlinker --wrap=trunc -Xlinker --wrap=floor -Xlinker --wrap=ceil -Xlinker --wrap=round -Xlinker --wrap=sincos -Xlinker --wrap=asin -Xlinker --wrap=acos -Xlinker --wrap=atan -Xlinker --wrap=sinh -Xlinker --wrap=cosh -Xlinker --wrap=tanh -Xlinker --wrap=asinh -Xlinker --wrap=acosh -Xlinker --wrap=atanh -Xlinker --wrap=exp2 -Xlinker --wrap=log2 -Xlinker --wrap=exp10 -Xlinker --wrap=log10 -Xlinker --wrap=pow -Xlinker --wrap=powint -Xlinker --wrap=hypot -Xlinker --wrap=cbrt -Xlinker --wrap=fmod -Xlinker --wrap=drem -Xlinker --wrap=remainder -Xlinker --wrap=remquo -Xlinker --wrap=expm1 -Xlinker --wrap=log1p -Xlinker --wrap=fma -Xlinker --wrap=__aeabi_lmul -Xlinker --wrap=__aeabi_fadd -Xlinker --wrap=__aeabi_fdiv -Xlinker --wrap=__aeabi_fmul -Xlinker --wrap=__aeabi_frsub -Xlinker --wrap=__aeabi_fsub -Xlinker --wrap=__aeabi_cfcmpeq -Xlinker --wrap=__aeabi_cfrcmple -Xlinker --wrap=__aeabi_cfcmple -Xlinker --wrap=__aeabi_fcmpeq -Xlinker --wrap=__aeabi_fcmplt -Xlinker --wrap=__aeabi_fcmple -Xlinker --wrap=__aeabi_fcmpge -Xlinker --wrap=__aeabi_fcmpgt -Xlinker --wrap=__aeabi_fcmpun -Xlinker --wrap=__aeabi_i2f -Xlinker --wrap=__aeabi_l2f -Xlinker --wrap=__aeabi_ui2f -Xlinker --wrap=__aeabi_ul2f -Xlinker --wrap=__aeabi_f2iz -Xlinker --wrap=__aeabi_f2lz -Xlinker --wrap=__aeabi_f2uiz -Xlinker --wrap=__aeabi_f2ulz -Xlinker --wrap=__aeabi_f2d -Xlinker --wrap=sqrtf -Xlinker --wrap=cosf -Xlinker --wrap=sinf -Xlinker --wrap=tanf -Xlinker --wrap=atan2f -Xlinker --wrap=expf -Xlinker --wrap=logf -Xlinker --wrap=ldexpf -Xlinker --wrap=copysignf -Xlinker --wrap=truncf -Xlinker --wrap=floorf -Xlinker --wrap=ceilf -Xlinker --wrap=roundf -Xlinker --wrap=sincosf -Xlinker --wrap=asinf -Xlinker --wrap=acosf -Xlinker --wrap=atanf -Xlinker --wrap=sinhf -Xlinker --wrap=coshf -Xlinker --wrap=tanhf -Xlinker --wrap=asinhf -Xlinker --wrap=acoshf -Xlinker --wrap=atanhf -Xlinker --wrap=exp2f -Xlinker --wrap=log2f -Xlinker --wrap=exp10f -Xlinker --wrap=log10f -Xlinker --wrap=powf -Xlinker --wrap=powintf -Xlinker --wrap=hypotf -Xlinker --wrap=cbrtf -Xlinker --wrap=fmodf -Xlinker --wrap=dremf -Xlinker --wrap=remainderf -Xlinker --wrap=remquof -Xlinker --wrap=expm1f -Xlinker --wrap=log1pf -Xlinker --wrap=fmaf -Xlinker --wrap=malloc -Xlinker --wrap=calloc -Xlinker --wrap=realloc -Xlinker --wrap=free -Xlinker --wrap=memcpy -Xlinker --wrap=memset -Xlinker --wrap=__aeabi_memcpy -Xlinker --wrap=__aeabi_memset -Xlinker --wrap=__aeabi_memcpy4 -Xlinker --wrap=__aeabi_memset4 -Xlinker --wrap=__aeabi_memcpy8 -Xlinker --wrap=__aeabi_memset8 -Xlinker -Map=picoprobe.elf.map -Xlinker --script=/mnt/d/u/pico/pico-sdk/src/rp2_common/pico_standard_link/memmap_default.ld -Xlinker -z -Xlinker max-page-size=4096 -Xlinker --gc-sections -Xlinker --wrap=printf -Xlinker --wrap=vprintf -Xlinker --wrap=puts -Xlinker --wrap=putchar -Xlinker --wrap=getchar CMakeFiles/picoprobe.dir/src/dap_util.c.obj CMakeFiles/picoprobe.dir/src/get_config.c.obj CMakeFiles/picoprobe.dir/src/led.c.obj CMakeFiles/picoprobe.dir/src/main.c.obj CMakeFiles/picoprobe.dir/src/misc_utils.c.obj CMakeFiles/picoprobe.dir/src/probe.c.obj CMakeFiles/picoprobe.dir/src/rtt_console.c.obj CMakeFiles/picoprobe.dir/src/sw_dp_pio.c.obj CMakeFiles/picoprobe.dir/src/sw_lock.c.obj CMakeFiles/picoprobe.dir/src/usb_descriptors.c.obj CMakeFiles/picoprobe.dir/CMSIS_5/CMSIS/DAP/Firmware/Source/DAP.c.obj CMakeFiles/picoprobe.dir/CMSIS_5/CMSIS/DAP/Firmware/Source/JTAG_DP.c.obj CMakeFiles/picoprobe.dir/CMSIS_5/CMSIS/DAP/Firmware/Source/DAP_vendor.c.obj CMakeFiles/picoprobe.dir/CMSIS_5/CMSIS/DAP/Firmware/Source/SWO.c.obj CMakeFiles/picoprobe.dir/src/daplink/daplink/util.c.obj CMakeFiles/picoprobe.dir/src/daplink/daplink/drag-n-drop/flash_manager.c.obj CMakeFiles/picoprobe.dir/src/daplink/daplink/interface/swd_host.c.obj CMakeFiles/picoprobe.dir/src/daplink/daplink/interface/target_flash.c.obj CMakeFiles/picoprobe.dir/src/daplink/daplink/settings/settings_rom_stub.c.obj CMakeFiles/picoprobe.dir/src/daplink/family/nordic/target_reset_nrf52.c.obj CMakeFiles/picoprobe.dir/src/daplink/family/nordic/nrf52/target.c.obj CMakeFiles/picoprobe.dir/src/daplink/target/target_board.c.obj CMakeFiles/picoprobe.dir/src/daplink/target/target_family.c.obj CMakeFiles/picoprobe.dir/src/daplink-pico/board/rp2040/pico.c.obj CMakeFiles/picoprobe.dir/src/daplink-pico/board/rp2040/pico_target_utils.c.obj CMakeFiles/picoprobe.dir/src/daplink-pico/board/rp2040/program_flash_generic.c.obj CMakeFiles/picoprobe.dir/src/daplink-pico/family/raspberry/rp2040/target_reset_rp2040.c.obj CMakeFiles/picoprobe.dir/src/cdc/cdc_uart.c.obj CMakeFiles/picoprobe.dir/src/cdc/cdc_debug.c.obj CMakeFiles/picoprobe.dir/src/net/net_glue.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/tinyusb/lib/networking/dhserver.c.obj CMakeFiles/picoprobe.dir/src/net/tinyusb/ncm_device_simple.c.obj CMakeFiles/picoprobe.dir/src/net/net_sysview.c.obj CMakeFiles/picoprobe.dir/src/net/net_echo.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/apps/lwiperf/lwiperf.c.obj CMakeFiles/picoprobe.dir/src/msc/msc_drive.c.obj CMakeFiles/picoprobe.dir/src/msc/msc_utils.c.obj CMakeFiles/picoprobe.dir/src/minIni/minIni.c.obj CMakeFiles/picoprobe.dir/src/minIni/minGlue-Flash.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/src/rp2_common/pico_lwip/lwip_freertos.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/src/rp2_common/pico_async_context/async_context_base.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/src/rp2_common/pico_platform/platform.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/init.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/def.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/dns.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/inet_chksum.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/ip.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/mem.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/memp.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/netif.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/pbuf.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/raw.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/stats.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/sys.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/altcp.c.obj CMakeFiles/picoprobe.dir/mnt/d/u/pico/pico-sdk/lib/lwip/src/core/altcp_alloc.c.obj CMakeFiles/picoprobe.dir/mnt/ [... cut ...]
	cmake -B $(BUILD_DIR) -G Ninja -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=1 -DPICO_BOARD=$(PICO_BOARD) $(CMAKE_FLAGS)
```

Invocation of the clang build (produces above error)
```
.PHONY: cmake-create-release-clang
cmake-create-release-clang:
	export PICO_TOOLCHAIN_PATH=~/bin/llvm-arm-none-eabi/bin
	cmake -B $(BUILD_DIR) -G Ninja -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=1 -DPICO_BOARD=$(PICO_BOARD) \
	         $(CMAKE_FLAGS) -DPICO_COMPILER=pico_arm_clang
```

Any ideas on this?

rgrr, 2023-08-02:
If one checks the generated obj file, then it seems, that clang release invents the symbol stdout out of the blue.

gcc undefined symbols in printf.c.obj:
```
00000000         *UND*  00000000 putchar
00000000         *UND*  00000000 __aeabi_dcmpun
00000000         *UND*  00000000 __aeabi_dcmpgt
00000000         *UND*  00000000 __aeabi_dcmplt
00000000         *UND*  00000000 __aeabi_i2d
00000000         *UND*  00000000 __aeabi_dmul
00000000         *UND*  00000000 __aeabi_dadd
00000000         *UND*  00000000 __aeabi_dsub
00000000         *UND*  00000000 __aeabi_d2iz
00000000         *UND*  00000000 __aeabi_ddiv
00000000         *UND*  00000000 __aeabi_dcmpge
00000000         *UND*  00000000 __aeabi_uidivmod
00000000         *UND*  00000000 __aeabi_idivmod
00000000         *UND*  00000000 __aeabi_uidiv
00000000         *UND*  00000000 __aeabi_idiv
00000000         *UND*  00000000 __aeabi_d2uiz
00000000         *UND*  00000000 __aeabi_ui2d
00000000         *UND*  00000000 __aeabi_uldivmod
```

clang:
```
00000000         *UND*  00000000 fputc
00000000         *UND*  00000000 stdout
00000000         *UND*  00000000 __aeabi_dcmpun
00000000         *UND*  00000000 __aeabi_dcmplt
00000000         *UND*  00000000 __aeabi_dcmpgt
00000000         *UND*  00000000 __aeabi_dsub
00000000         *UND*  00000000 __aeabi_memset4
00000000         *UND*  00000000 __aeabi_d2iz
00000000         *UND*  00000000 __aeabi_i2d
00000000         *UND*  00000000 __aeabi_dmul
00000000         *UND*  00000000 __aeabi_d2uiz
00000000         *UND*  00000000 __aeabi_ui2d
00000000         *UND*  00000000 __aeabi_dcmple
00000000         *UND*  00000000 __aeabi_dcmpeq
00000000         *UND*  00000000 __aeabi_idiv
00000000         *UND*  00000000 __aeabi_idivmod
00000000         *UND*  00000000 __aeabi_memset
00000000         *UND*  00000000 __aeabi_dadd
00000000         *UND*  00000000 __aeabi_ddiv
00000000         *UND*  00000000 __aeabi_dcmpge
00000000         *UND*  00000000 __aeabi_uidivmod
```

Wondering if this actually is an issue of picolib or compiler-rt?

kilograham, 2023-08-02:
Yeah, from the release notes for 1.5.1

> Experimental support is provided for compiling with Clang. As an example, you can build with the
LLVM Embedded Toolchain for Arm, noting however that currently only version 14.0.0 works, as later versions use picolib rather than newlib.

rgrr, 2023-08-02:
Thanks Graham for pointing me to the release notes...

Tried clang 14.0.0 as suggested and almost everything works out of the box (except some minor glitches with static_assert()).

Anything I can do to help making the SDK work with picolib and later clangs?

kilograham, 2023-08-03:
Yeah, i didn't have any time to look at it in detail, but the area of integration with the C-library and the somewhat wide distribution of such points within the SDK, suggests to me that we should factor that adapter layer out into a single library perhaps, and allow plugging that (or via #ifdef) 

But even short of that, just prototyping something that works with picolib, even if it isn't abstracted out

geurtv, 2023-08-04:
I did some tests with clang 16.0.0 (on Windows) and a c++ project. After some small changes and resolving undefined symbols everything just works:

1. `pico_arm_clang.cmake`:  removed `--sysroot`
The --sysroot path wasn't correct for clang-16 but the option also seems unnecessary. Everything builds just fine without it.
(I think `--target=armv6m-none-eabi` sets everything up)

2. undefined `stdout`
Simply added to source file: `std::FILE* const stdout = nullptr;`. I assume this is good enough and it's not actually used (in my code anyway). `printf`, for instance, seems to work without issue.

3. undefined `extern "C" void __cxxabiv1::__cxa_pure_virtual();`
Implemented in source file: calls panic()

4. undefined `__heap_start`
Added `__heap_start = .;` to memmap_default.ld before `__end__`

5. undefined `__heap_end`
Added `__heap_end = .;` to memmap_default.ld before `__scratch_x_start__`

6. undefined `__atomic_load_[1/2/4]` and `__atomic_store_[1/2/4]`
gcc provides implementations for these while clang does not. I can provide (c++) implementations. 

I still have 4 linker warnings all similar to:
```
ld.lld: warning: section type mismatch for .ram_vector_table
>>> .../pico_runtime/runtime.c.obj:(.ram_vector_table): SHT_PROGBITS
>>> output section .ram_vector_table: SHT_NOBITS
```
The others are for `.heap`, `.stack` and `.stack1`. Not sure what to do with them.

geurtv, 2023-08-04:
@kilograham yes they were with release-16.0.0 from [LLVM Embedded Toolchain for Arm](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm).

rgrr, 2023-08-09:
I can almost confirm what @geurtv wrote.  My path was a little bit easier, because (my guess) my program is "C".

cmake invocation:

```
export PICO_TOOLCHAIN_PATH=~/bin/llvm-arm-none-eabi/bin
cmake -B $(BUILD_DIR) -G Ninja -DCMAKE_BUILD_TYPE=Release -DPICO_BOARD=$(PICO_BOARD) \
	         $(CMAKE_FLAGS) -DPICO_COMPILER=pico_arm_clang
```

Things I had to fix:

1. clang throws some errors about static asserts without messages.  Added those messages
2. clang has a different opinion about "-Wformat" because it treats uint32_t differently than gcc (or the typedef is different, don't know).  Added "-Wno-format"
3. clang does not know the "optimize" attribute which is used somewhere in a library.  Added "-Wno-unknown-attributes"
4. note: `--sysroot` pointed to the correct location, so no changes required to `pico_arm_clang.cmake`
5. `stdout`: I had to add `FILE *const stdout = NULL;` because "C"

That was it.  Compiles and executes now fine with debug/release.

I compiled with a homebrewed version of the LLVM Embedded Toolchain which resembles more or less version 17 before they removed the *.cfg files.

geurtv, 2023-08-09:
@rgrr I assume you don't use heap allocations, so no `malloc()` and `free()`? Otherwise I would expect it to complain about `__heap_start` and `__heap_end` as well. 

Also, just checked, on gcc `uint32_t` is `unsigned long` and on clang it's `unsigned int`, hence the `printf` warnings. BTW, the portable way to `printf` `uint32_t` is to use `PRIu32` (from inttypes.h), so for instance:

```c
void print_values(uint32_t value1, int32_t value2)
{
	printf("value1=%" PRIu32", value2=%" PRIi32"\n", value1, value2);
}
```

kilograham, 2023-08-09:
>     1. clang throws some errors about static asserts without messages.  Added those messages
> 
>     3. clang does not know the "optimize" attribute which is used somewhere in a library.  Added "-Wno-unknown-attributes"

Were these in pico-sdk/pico-examples; they should have all been fixed.

rgrr, 2023-08-09:
:
> Were these in pico-sdk/pico-examples; they should have all been fixed.

Don't worry... it was somewhere in my daplink library copy.  The only warning concerning RP2040 came from FreeRTOS which complains about `warning: unused function 'prvGetAllEventGroupBits'` in port.c

rgrr, 2023-08-09:
> @rgrr I assume you don't use heap allocations, so no `malloc()` and `free()`? Otherwise I would expect it to complain about `__heap_start` and `__heap_end` as well.

Yes, you are right.  My actual application does no malloc()/free().  Only (pseudo) dynamic things are done in FreeRTOS and lwIP.

> Also, just checked, on gcc `uint32_t` is `unsigned long` and on clang it's `unsigned int`, hence the `printf` warnings. BTW, the portable way to `printf` `uint32_t` is to use `PRIu32` (from inttypes.h), so for instance:

Gosh, that's really unfortunate.

> ```c
> void print_values(uint32_t value1, int32_t value2)
> {
> 	printf("value1=%" PRIu32", value2=%" PRIi32"\n", value1, value2);
> }
> ```

Oh, that's new to me.  I agree with Graham, that this is rather annoying and normally I change the parameters via casting as well.

voltur01, 2023-11-21:
Hi,

I noticed the discussion about `picolibc` vs `newlib` above, so I want to point put that the [LLVM embedded toolchain for Arm](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm) added experimental set of multilib variants based on `newlib` 4.1.0, see the [dev drop](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases/tag/preview-18.0.0-devdrop0).

If this proves useful, `newlib` set of libraries will be included into every release, keeping `picolibc` the default.

Please report any feedback back to the [LLVM embedded toolchain for Arm](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/issues) project.

kilograham, 2023-11-21:
> Hi,
>
> I noticed the discussion about `picolibc` vs `newlib` above, so I want to point put that the [LLVM embedded toolchain for Arm](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm) [...]

excellent, thanks

kilograham, 2024-08-09:
fixed in SDK 2.0.0

closed 2024-08-09 by kilograham