raspberrypi/pico-sdk #1472: BT LE scan while advertising issue

matsobdev, 2023-08-10:
Program doing advertisements and passively scanning at the same time with long window (equal interval = 100% scan duty), ex. 0x4000 = 10.24 s stalls for random time (seconds) from time to time (minutes). At stall HCI communication stops. Not only Pico W advertising, but couple beacons at Android phone. With very short intervals stalls are not noticeable. Maybe by shorter time periods is not observable. With or without whitelist - doesn't matter. [gap_le_scan.zip](https://github.com/raspberrypi/pico-sdk/files/12317384/gap_le_scan.zip) to work with. Tested on vanilla 1.5.1 and modified (run from RAM, added ROSC from extras, running from XOSC and turned off other clock sources) - same results. LED stops blinking when it stalls.

peterharperuk, 2024-10-24:
Your example doesn't seem to be stalling for me. How easy is this to reproduce?

matsobdev, 2024-11-12:
Couple of minutest of waiting. Maybe it easier to start one first, wait couple minutes and start other, but it's more likely to be a coincidence. Upper one just stalled when started to record, and later lower one did 30ish seconds later:

[image]

There were no extra advertisements present. But with external (phone) beacons Pico W's reactions is the same. But from Android phone, when combined advertisements are about 3-4 times less frequent (than Pico's 20 ms interval), it is harder to stall it. UF2s used:
[UF2.zip](https://github.com/user-attachments/files/17716998/UF2.zip). There is 2.0.0 and 1.5.1, but it acts the same. I don't know, maybe my SDK copies are wired up or something, if you'd like to provide your binaries, I can check them.