raspberrypi/pico-sdk #1478: Core changes while in cyw43_driver using FreeRTOS SMP

The-Michael-R, 2023-08-19:
Sorry, I don't have a debugger at hand, so all is done via debug-print.

I try to run my Pico_W with FreeRTOS-SMP and WIFI and follow loosely the example. Everything is fine, except that the core switches while within the cyw43_driver. A small change in `cyw43_driver.c` to see what's going on:
```c
// ...
static void cyw43_do_poll(async_context_t *context, __unused async_when_pending_worker_t *worker) {
#ifndef NDEBUG
    printf("--------------------------->4<---------------------\n");
    printf("async_context_core_num(cyw43_async_context) = %d\n", async_context_core_num(cyw43_async_context));
    printf("get_core_num() = %d\n", get_core_num());
    printf("get_core_num() = %d\n", get_core_num());
    printf("get_core_num() = %d\n", get_core_num());
    printf("get_core_num() = %d\n", get_core_num());
    assert(get_core_num() == async_context_core_num(cyw43_async_context));
#endif
// ---
```
Produces the following outputs:
```
Version: 7.95.49 (2271bb6 CY) CRC: b7a28ef3 Date: Mon 2021-11-29 22:50:27 PST Ucode Ver: 1043.2162 FWID 01-c51d9400
cyw43 loaded ok, mac xx:xx:xx:xx:xx:xx 
API: 12.2
Data: RaspberryPi.PicoW
Compiler: 1.29.4 
ClmImport: 1.47.1
Customization: v5 22/06/24 
Creation: 2022-06-24 06:55:08
[...]
--------------------------->4<---------------------
async_context_core_num(cyw43_async_context) = 0
get_core_num() = 1 
get_core_num() = 1 
get_core_num() = 0 
get_core_num() = 0 
--------------------------->3<---------------------
--------------------------->4<---------------------
async_context_core_num(cyw43_async_context) = 0
get_core_num() = 1 
get_core_num() = 1 
get_core_num() = 1 
get_core_num() = 1
assertion "get_core_num() == async_context_core_num(cyw43_async_context)" failed: file "..."
```
First thought was I missed the `vTaskCoreAffinitySet()` but it is there:

```c
#define WLAN_PRIORITY (tskIDLE_PRIORITY + 1UL)

void WlanRtosInit(void)
{
    TaskHandle_t tTaskHandle;

    xTaskCreate(vTaskFunction,
                "WLAN",
                configMINIMAL_STACK_SIZE,
                NULL,
                WLAN_PRIORITY,
                &tTaskHandle);

    vTaskCoreAffinitySet(tTaskHandle, 1);  /* <---- Here the affinity is set to core 0  */
}

void vTaskFunction(void *pvParameters)
{
    if (cyw43_arch_init_with_country(CYW43_COUNTRY_GERMANY))
    {
        printf("failed to initialise\n");
        eRetVal = err_error;
    }
    else
    {
        cyw43_arch_enable_sta_mode();
        if (cyw43_arch_wifi_connect_timeout_ms(WLAN_SSID, WLAN_PASSWORD, CYW43_AUTH_WPA2_AES_PSK, 30000))
        // ...
    }
}
```
Note that I see the assert fires at all four places within `cyw43_driver.c` at random. And the `vTaskFunction()` runs reliable on core 0.

kilograham, 2023-08-20:
> follow loosely the example

you'll have to be a bit more specific... what libraries are you linking in your `CMakeLists.txt`, particularly are you sure you're using the FreeRTOS version of pico_cyw43 stuff

The-Michael-R, 2023-08-20:
Sorry for not provide all info from the beginning. But by collecting all information I freshly updated the local FreeRTOS and pico-sdk reprositories and the problem is gone. And everything works fine now.

Anyway, thanks for your support! :+1:

closed 2023-08-20 by The-Michael-R