raspberrypi/pico-sdk #1479: Instability when exiting btstack run loop

josh26turner, 2023-08-19:
I'm not sure if this issue should sit in this repo or BTStack repo.

I have an application for the Pico W that will periodically take a measurement and send the value over a Bluetooth notifcation to a connected device. As it's battery powered, after sending a notification, I run the function `btstack_run_loop_trigger_exit()`, and have a loop that looks (roughly) like this:

```c
while (true) {
    printf("Starting btstack\n");
    btstack_run_loop_execute();
    printf("Exited btstack\n");

    printf("Sleeping for %ds\n", POST_MEASUREMENT_SLEEP_TIME_MS / 1000);
    sleep_ms(POST_MEASUREMENT_SLEEP_TIME_MS);

    ## Less important, queuing up work for when btstack starts again
    if (le_notification_enabled)
    {
        btstack_run_loop_set_timer(&sensor_reading_timer, 0);
        btstack_run_loop_add_timer(&sensor_reading_timer);
    }
}
```

Which seems to work for the first few measurement, producing logs that look like this (the "`Exiting btstack`" log line is printed immediately before calling `btstack_run_loop_trigger_exit()`):
```log
Starting btstack
BTstack up and running on 28:CD:C1:09:3E:D0.
Enabling notifications
Requesting reading
Sensor values read
Sending notification to connected device
Exiting btstack
Exited btstack
Sleeping for 10s
Starting btstack
Requesting reading
Sensor values read
Sending notification to connected device
Exiting btstack
Exited btstack
Sleeping for 10s
Starting btstack
Requesting reading
Sensor values read
Sending notification to connected device
Exiting btstack
Exited btstack
```

However, after a few successful attempts (not consistent in number), the exit command will *jump* back code when exiting the run loop:
```log
Starting btstack
Requesting reading
Sensor values read
Sending notification to connected device
Exiting btstack
Exited btstack
Sleeping for 10s
Requesting reading
Sensor values read
Sending notification to connected device
Exiting btstack
Starting btstack
```

You can see that immediately following the "`Exiting btstack`" log line is the line "`Starting btstack`". At which point the program hangs (as I haven't queued up any work for btstack to do). It can be worked around by checking if a sleep is required an exiting if so, but that is not a long term solution.

If required, I can add some example code to run and replicate.

peterharperuk, 2023-09-05:
I think we'd need to see the full code. I don't think we expect the code to be used like that.

peterharperuk, 2025-05-01:
This might fix the problems stopping bluetooth https://github.com/raspberrypi/pico-sdk/pull/2446
I've pushed an example here which seems to work https://github.com/raspberrypi/pico-examples/pull/644

kilograham, 2025-07-21:
moving to 2.2.0 as it seems possible this might already be fixed in `develop`?