raspberrypi/pico-sdk #1482: Compilation on MacOs - M2

HouseSimOn, 2023-08-26:
I've performed steps from https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf to install compiler:
```
$ brew install cmake
$ brew tap ArmMbed/homebrew-formulae
$ brew install arm-none-eabi-gcc
```
After that I tried to compile project and I got following error:
`arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory macOS`

Then I uninstalled `arm-none-eabi-gcc` using brew and installed `brew install --cask gcc-arm-embedded`. Now compilation works, but I see that compiler identification is `AppleClang` rather than `arm-none-eabi-`. 

Configuring make I see correct compilers:
```
% cmake -G "Unix Makefiles" ..
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 12.3.1
-- The CXX compiler identification is GNU 12.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/homebrew/bin/arm-none-eabi-gcc
```

But during compilation I see `AppleClang`:
```
% make 
[  1%] Creating directories for 'ELF2UF2Build'
[  2%] No download step for 'ELF2UF2Build'
[  3%] No update step for 'ELF2UF2Build'
[  4%] No patch step for 'ELF2UF2Build'
[  5%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is AppleClang 14.0.3.14030022
-- The CXX compiler identification is AppleClang 14.0.3.14030022
```

Why there is info about `AppleClang` compiler?

kilograham, 2023-08-27:
Because tools like `elf2uf2` and `pioasm` are compiled from source using the native compiler

lurch, 2023-08-27:
ping @aallan in case the instructions in the Getting Started PDF need updating for M2 Macs.

stefannilsson, 2023-09-22:
Thanks a lot, @timlukins ! Worked great on my M1 too.

GoldeggCoder, 2023-10-06:
Thanks alot! Worked also on my M2 machine!!! Searched for hours. Got it now

allthingsclowd, 2023-10-10:
Thank you - this solved an error I was getting on my Apple M2 Max.

"arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory"

Got this after following the "Getting started Guide"

aallan, 2023-10-10:
Okay, transferring this to the pico-feedback repo as a documentation problem.