raspberrypi/pico-sdk #1483: CMake deprecation warning displayed when pioasm is built

kripton, 2023-08-27:
I lately compiled one of my projects using cmake 3.27.4 and was greeted with this deprecation notice:
```
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
```

I tracked this down to the CMakeLists.txt file of pioasm:
https://github.com/raspberrypi/pico-sdk/blob/master/tools/pioasm/CMakeLists.txt#L1

The options I see:
* Either update the minimum version required `VERSION 3.5`
* Add a newer version that it has been also tested with so CMake is aware that the old, compatibility behaviour is not longer required: `VERSION 3.4...3.27` (or some other version between 3.5 and 3.27)

I'll create a PR to implement option 2 since pioasm builds fine for me using cmake 3.27.4. If you want to specify an older version here that's fine for me as well

kripton, 2023-08-27:
See also: https://discourse.cmake.org/t/how-to-fix-cmake-minimum-required-deprecation-warning/2487

lurch, 2023-08-27:
Fixed by #1477 ?

kripton, 2023-08-28:
Indeed, fixed already by #1477. I just checked the latest release, not the develop branch.

closed 2023-08-28 by kripton