raspberrypi/pico-sdk #1485: assert(rt->alarm_id == id) fails for repeating timers on two cores

MrtzGtz, 2023-08-27:
I created a repeating timer independently on each core with `alarm_pool_create` and `alarm_pool_add_repeating_timer_us` with different hardware alarms for each alarm_pool.

When using small values for the `delay_us` (like 40) `assert(rt->alarm_id == id);` fails in `repeating_timer_callback`.

I created a project to reproduce the error: [pico_timer_error](https://github.com/MrtzGtz/pico_timer_error).

I already researched the problem and only found others which had errors with the timer, like #737 and [Pico Forum: add_alarm_in_ms - assertion?](https://forums.raspberrypi.com/viewtopic.php?t=348922&)

kilograham, 2023-08-28:
thanks for the repo, that will be a big help!

kilograham, 2023-08-29:
ah, that assertion is actually incorrect - it breaks if the repeating timer has a very short period, and fires before creating the timer is fully complete
[image]

closed 2023-08-29 by kilograham