raspberrypi/pico-sdk #1486: Problem With Building: Multiple Definitions Even If I Use #ifndef

MertKalkanci, 2023-08-27:
Hey there here is the code gives problems:
shared.h:
```c
#include "pico/stdlib.h"
#include "nrf24l01/nrf24_driver.h"

#ifndef CLIENT_ID
#define CLIENT_ID

unsigned int clientId_; //modify per client CLIENT ID
unsigned int clientId; //modify per client CLIENT ID

#endif

#ifndef PAYLOAD_STRUCT
#define PAYLOAD_STRUCT

typedef struct payloads { 
      uint8_t moistures[200];
      uint8_t last_device_id;
      } payload_struct;

#endif

#ifndef GLOBAL_PINS_DEF
#define GLOBAL_PINS_DEF

pin_manager_t global_pins;

#endif
```
shared.c
```c
#include "pico/stdlib.h"
#include "shared.h"
#include "nrf24l01/nrf24_driver.h"

#ifndef PIPES_VAL
#define PIPES_VAL

unsigned int clientId_ = 0; //modify per client CLIENT ID
unsigned int clientId = 0; //modify per client CLIENT ID

#endif

#ifndef PINS_VAL    
#define PINS_VAL

pin_manager_t global_pins = { //DO NOT MODIFY
    .sck = 10,
    .copi = 11, 
    .cipo = 12, 
    .csn = 13, 
    .ce = 14 
};

#endif
```

Errors:
```
[ 12%] Linking CXX executable akasu-humidity-unit-0.elf
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/receive.c.obj:(.bss.global_pins+0x0): multiple definition of `global_pins'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.global_pins+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/receive.c.obj:(.bss.receivingDeviceId+0x0): multiple definition of `receivingDeviceId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.receivingDeviceId+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/receive.c.obj:(.bss.deviceId+0x0): multiple definition of `deviceId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.deviceId+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/receive.c.obj:(.bss.clientId_+0x0): multiple definition of `clientId_'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.clientId_+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/receive.c.obj:(.bss.clientId+0x0): multiple definition of `clientId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.clientId+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/send.c.obj:(.bss.global_pins+0x0): multiple definition of `global_pins'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.global_pins+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/send.c.obj:(.bss.receivingDeviceId+0x0): multiple definition of `receivingDeviceId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.receivingDeviceId+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/send.c.obj:(.bss.deviceId+0x0): multiple definition of `deviceId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.deviceId+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/send.c.obj:(.bss.clientId_+0x0): multiple definition of `clientId_'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.clientId_+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/send.c.obj:(.bss.clientId+0x0): multiple definition of `clientId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.clientId+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/shared.c.obj:(.data.global_pins+0x0): multiple definition of `global_pins'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.global_pins+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/shared.c.obj:(.data.receivingDeviceId+0x0): multiple definition of `receivingDeviceId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.receivingDeviceId+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/shared.c.obj:(.bss.deviceId+0x0): multiple definition of `deviceId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.deviceId+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/shared.c.obj:(.bss.clientId_+0x0): multiple definition of `clientId_'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.clientId_+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/akasu-humidity-unit-0.dir/shared.c.obj:(.bss.clientId+0x0): multiple definition of `clientId'; CMakeFiles/akasu-humidity-unit-0.dir/main.c.obj:(.bss.clientId+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/akasu-humidity-unit-0.dir/build.make:1319: akasu-humidity-unit-0.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1578: CMakeFiles/akasu-humidity-unit-0.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

There isn't any problem caused by the nrf24l01 library but whenever i call this shared.h file I get multiple definitions error while using "make".

2 Days ago everythign was fine on Ubuntu 20.04 LTS now im at Ubuntu 22.04 LTS having the same errors when i switched to Debain 12 too so I think that it's not an os issue may be a gcc or something related with it ?

commands I runned to instal gcc & other stuff:
```bash

cd ~/
wget https://raw.githubusercontent.com/raspberrypi/pico-setup/master/pico_setup.sh
sudo bash ./pico_setup.sh
export PICO_SDK_PATH=~/pico/pico-sdk/
sudo reboot

```

just got an error at visual studio section and copy pasted code comes 

in this section:
```
sudo raspi-config nonint do_serial 2
```
after it  got an error like:
```
sudo: raspi-config: command not found
```

Also I tried something too:

shared.h:
```c
#include "pico/stdlib.h"
#include "nrf24l01/nrf24_driver.h"

#ifndef CLIENT_ID
#define CLIENT_ID 1

unsigned int clientId_; //modify per client CLIENT ID
unsigned int clientId; //modify per client CLIENT ID

#endif

#ifndef PAYLOAD_STRUCT
#define PAYLOAD_STRUCT 1

typedef struct payloads { 
      uint8_t moistures[200];
      uint8_t last_device_id;
      } payload_struct;

#endif

#ifndef GLOBAL_PINS_DEF
#define GLOBAL_PINS_DEF

pin_manager_t global_pins;

#endif
```
shared.c
```c
#include "pico/stdlib.h"
#include "shared.h"
#include "nrf24l01/nrf24_driver.h"

#ifndef PIPES_VAL
#define PIPES_VAL 1

unsigned int clientId_ = 0; //modify per client CLIENT ID
unsigned int clientId = 0; //modify per client CLIENT ID

#endif

#ifndef PINS_VAL    
#define PINS_VAL 1

pin_manager_t global_pins = { //DO NOT MODIFY
    .sck = 10,
    .copi = 11, 
    .cipo = 12, 
    .csn = 13, 
    .ce = 14 
};

#endif
```

ALSO  I tried  
```c
 #pragma once 
```

it didn't worked too.

In the error giving C files I've just done:

```c
#include "shared.h" 
```

What should I do now ?

MertKalkanci, 2023-08-27:
Also here is the cmake file 
```cmake

cmake_minimum_required(VERSION 3.12) # minimum cmake version

include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake) # import the pico sdk

project(akasu-humidity-unit-0 C CXX ASM) 

set(CMAKE_C_STANDARD 11) # set the C standard
set(CMAKE_CXX_STANDARD 17) # set the C++ standard

pico_sdk_init() # initialize the sdk

add_subdirectory(nrf24l01) # add the nrf24l01 communication library

add_executable(${PROJECT_NAME} # create the executable
    main.c
)

add_library(moisture INTERFACE) # create humidity sensor library
target_sources(moisture 
    INTERFACE
      ${CMAKE_CURRENT_LIST_DIR}/moisture.c
)

add_library(receive INTERFACE) # create reciever library
target_sources(receive 
    INTERFACE
      ${CMAKE_CURRENT_LIST_DIR}/receive.c
)

add_library(send INTERFACE) # create sender library
target_sources(send 
    INTERFACE
      ${CMAKE_CURRENT_LIST_DIR}/send.c
)

add_library(shared INTERFACE) # create shared library
target_sources(shared 
    INTERFACE
      ${CMAKE_CURRENT_LIST_DIR}/shared.c
)

target_link_libraries(${PROJECT_NAME} # link the libraries to the executable
    hardware_adc
    hardware_gpio
    pico_stdlib 
    nrf24_driver
    moisture
    receive
    send
    shared
)

# Enable usb output, disable uart output
pico_enable_stdio_usb(${PROJECT_NAME} 1)
pico_enable_stdio_uart(${PROJECT_NAME} 0)

pico_add_extra_outputs(${PROJECT_NAME}) # add extra outputs like map/bin/hex/uf2 files

```

MertKalkanci, 2023-08-27:
Probably because of the
```c
typedef struct payloads { 
      uint8_t moistures[200];
      uint8_t last_device_id;
      } payload_struct;
```
section

lurch, 2023-08-27:
> whenever i call this shared.h file I get multiple definitions error while using "make".

That's because you've declared the same variable-names in both the header file (`shared.h`) and also the implementation file (`shared.c`). Note that the `#include` directive is handled in C by the pre-processor rather than the compiler, so what you've effectively got is:
```c
#include "pico/stdlib.h"
// start of included shared.h content
#include "pico/stdlib.h"
#include "nrf24l01/nrf24_driver.h"

#ifndef CLIENT_ID
#define CLIENT_ID

unsigned int clientId_; //modify per client CLIENT ID
unsigned int clientId; //modify per client CLIENT ID

#endif

#ifndef PAYLOAD_STRUCT
#define PAYLOAD_STRUCT

typedef struct payloads { 
      uint8_t moistures[200];
      uint8_t last_device_id;
      } payload_struct;

#endif

#ifndef GLOBAL_PINS_DEF
#define GLOBAL_PINS_DEF

pin_manager_t global_pins;

#endif
// end of included shared.h content
#include "nrf24l01/nrf24_driver.h"

#ifndef PIPES_VAL
#define PIPES_VAL

unsigned int clientId_ = 0; //modify per client CLIENT ID
unsigned int clientId = 0; //modify per client CLIENT ID

#endif

#ifndef PINS_VAL    
#define PINS_VAL

pin_manager_t global_pins = { //DO NOT MODIFY
    .sck = 10,
    .copi = 11, 
    .cipo = 12, 
    .csn = 13, 
    .ce = 14 
};

#endif
```

This is a generic 'C' programming problem and not specific to pico-sdk, so you should ask for further help at https://forums.raspberrypi.com/viewforum.php?f=143 rather than on GitHub.

MertKalkanci, 2023-08-28:
Thank you for your reply, it didn't helped so i moved my typedef struct to all fiiles uses them seperately and made some changes for it also pin manager and clientId's contuinued giving errors so I made them static.

Thanks for reply

closed 2023-08-28 by MertKalkanci