raspberrypi/pico-sdk #1491: Missed the 0.80V

hndrbrm, 2023-09-02:
https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/rp2_common/hardware_vreg/include/hardware/vreg.h#L26

I want to set the voltage to 0.80V but realize its not in there.

The datasheet says (Table 189. VREG Register):
`0000 to 0101 - 0.80V`

lurch, 2023-09-02:
Yup, that's also what the comments in https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/hardware_regs/include/hardware/regs/vreg_and_chip_reset.h say. I guess this also means that the `VREG_VOLTAGE_MIN` in https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_vreg/include/hardware/vreg.h needs updating too.

matsobdev, 2023-09-03:
You might not go anywhere with 0.80V anyway - probably won't boot.