raspberrypi/pico-sdk #1492: watchdog_get_count() never decrements

aallan, 2023-09-05:
An issue was raised on pico-feedback around the watchdog_get_count() never decrementing, see raspberrypi/pico-feedback#349. This was an issue with PlatformIO and the @earlephilhower Arduino core. 

However, @JamesH65 mentioned that it is also present in the SDK, so raising here.

kilograham, 2023-09-07:
This appears to be an RP2040 erratum, which will need to be documented

kilograham, 2023-12-14:
we should fix the documentation to say it doesnt work

kilograham, 2024-08-09:
This is documented in the code; happily it even works on RP2350

(note function has a new preferred name `watchdog_get_time_remaining_ms()`

cpottle9, 2025-04-30:
This issue needs to be documented in other places.
Many people do not read the SDK source code.

Specifically in:
- the RP2040 datasheet as an errata and additional detail in the description of the watchdog CTRL register.
- online documentation [hardware_watchdog](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_watchdog)

kilograham, 2025-05-01:
still needs an errata - also the old function name should probably link to the new function name in the docs (which documents the limitation)