raspberrypi/pico-sdk #1493: fatal error: stdlib.h: No such file or directory

ekaitz-zarraga, 2023-09-05:
Hi,

I'm following the README for the minimal setup and I'm unable to build the project.
This is my `CMakeLists.txt`

``` cmake
cmake_minimum_required(VERSION 3.13)

# initialize the SDK based on PICO_SDK_PATH
# note: this must happen before project()
set(PICO_SDK_PATH ../../pico-sdk)
include("pico_sdk_import.cmake")

project(my_project)

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

# rest of your project
add_executable(hello_world
    hello_world.c
)

# Add pico_stdlib library which aggregates commonly used features
target_link_libraries(hello_world pico_stdlib)

# create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(hello_world)
```

The cmake step works perfectly fine but I'm unable to `make`.
I'm on Guix System. And I have a proper `arm-none-eaby-toolchain` already set up.

This is what I see:

```
~/projects/rp2040/TESTS/build [env]$ cmake ..
PICO_SDK_PATH is /home/Ekaitz/projects/rp2040/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /gnu/store/xc1hs9952fi3mxf32wli9mpyjfj98g2x-profile/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /gnu/store/xc1hs9952fi3mxf32wli9mpyjfj98g2x-profile/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /gnu/store/xc1hs9952fi3mxf32wli9mpyjfj98g2x-profile/bin/arm-none-eabi-gcc
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/Ekaitz/projects/rp2040/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /home/Ekaitz/.guix-profile/bin/python3.10 (found version "3.10.7") found components: Interpreter
TinyUSB available at /home/Ekaitz/projects/rp2040/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/Ekaitz/projects/rp2040/pico-sdk/lib/btstack
cyw43-driver available at /home/Ekaitz/projects/rp2040/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/Ekaitz/projects/rp2040/pico-sdk/lib/lwip
mbedtls available at /home/Ekaitz/projects/rp2040/pico-sdk/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/Ekaitz/projects/rp2040/TESTS/build
Ekaitz@tuxedo ~/projects/rp2040/TESTS/build [env]$ make
[... 67 lines cut ...]
[ 68%] Building ASM object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_double/double_v1_rom_shim.S.obj
[ 70%] Building ASM object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_int64_ops/pico_int64_ops_aeabi.S.obj
[ 71%] Building ASM object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_float/float_aeabi.S.obj
[ 73%] Building C object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_float/float_init_rom.c.obj
[ 74%] Building C object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_float/float_math.c.obj
[ 76%] Building ASM object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_float/float_v1_rom_shim.S.obj
[ 77%] Building C object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c.obj
[ 79%] Building ASM object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_mem_ops/mem_ops_aeabi.S.obj
[ 80%] Building ASM object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_standard_link/crt0.S.obj
[ 82%] Building CXX object CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj
In file included from /home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:11:0:
/gnu/store/xc1hs9952fi3mxf32wli9mpyjfj98g2x-profile/arm-none-eabi/include/c++/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
               ^~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/hello_world.dir/build.make:653: CMakeFiles/hello_world.dir/home/Ekaitz/projects/rp2040/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1509: CMakeFiles/hello_world.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

Any ideas of what can be wrong?

Guix is a weird distribution sometimes because it doesn't follow the File Hierarchy Standard, but it sets all the `$CROSS_*` variables so `cmake` should be able to find them.

Thanks!

kilograham, 2023-09-05:
Better question for the forums https://forums.raspberrypi.com/viewforum.php?f=143

looks like your compiler install does not have newlib tho

closed 2023-09-05 by kilograham

ekaitz-zarraga, 2023-09-05:
> looks like your compiler install does not have newlib tho

It does have it! Maybe it's not able to find it?
This is weird. I'll try in the forums. Thanks