raspberrypi/pico-sdk #1496: BLE Services and Chars don't change between examples

cwillforeal, 2023-09-08:
Hi,

When I was testing out the BLE examples I noticed that the BLE services and characteristics wouldn't change. The device name would update but none of the attributes would update.
For example if I removed the temperature service/char it would still show up when I connected to the device.
I then tried the gatt_streamer_server example, the name changed to "LE_Streamer" but the temperature remained as the only service discovered.
This continues when I power cycle the chip to upload new uf2 files.
Is there some sort of "clean" that needs to happen with the BLE chip? It almost seems like it won't update its gatt database or something.
The SDK I'm using is the latest main branch.

sonnny, 2023-09-09:
If you are connecting through a phone, try to go to phone settings, under bluetooth connection click on picow, forget connection, then try to pair again see it that makes a difference.

Sonny

cwillforeal, 2023-09-13:
Hi thanks for the response. Unfortunately I can't see anything under BLE connections for the picow. My guess is that since the device isn't pairing so it won't show up in the bluetooth settings.
Something strange is happening with the phone though, it correctly discovers all the services after some time not being connected. I'll test using a computer and see if I get anything different.

peterharperuk, 2023-10-03:
This doesn't seem to happen for me

cwillforeal, 2023-10-05:
Looks like it's something funky going on with my phone. I tried using the NRF connect software with the nrf dongle and it worked just fine.

closed 2023-10-05 by cwillforeal