raspberrypi/pico-sdk #1507: Enable I2C to answer to read option in a slave mode

igorbezsmertnyi, 2023-09-11:
Hi there 👋

I'm trying to use RPi pico as an i2c slave and connect it to the Nvidia Jetson Nano board via an i2c bus. Currently, I'm facing a problem because for some reason Nvidia Jetson requires `-r`(read flag) when I try to run the `i2cdetect` command, `(i2cdetect -y -r 0)`. The other devices work fine, but pico stacked after sending `ack` and pulled SCL to 0. That's how it looks on the analyzer screen

[image]

And that's how it should look like (other device)

[image]

Also, I did a small research and found this [piece of code](https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/rp2040/hardware_regs/include/hardware/regs/i2c.h#L1222) in pico-sdk. But those lines aren't used anywhere.

If somebody had the same issues as me, please help me modify the pico-sdk in a proper way to enable it to proceed connection process.

lurch, 2023-09-12:
I don't know if it helps, but have you looked at #1205 and https://github.com/raspberrypi/pico-examples/tree/master/i2c/slave_mem_i2c ?

progvault, 2024-03-15:
@igorbezsmertnyi - the TX FIFO is probably empty and the slave is stalling the bus by pulling SCL low (i.e. stretching the clock).  Look at your code and make sure you keep feeding the FIFO with data.